WebControlCNC/WebControl #12: Can't select port until groundcontrol.ini is imported

madgrizzle, 2019-03-19:
There have been reports that you can't do anything (e.g., select port for controller) unless you import your groundcontrol.ini file.  I have been unable to duplicate this issue to date.

If you have a clean install (first time running) and you go to Settings->Maslow Settings, the port entry should be blank.  If you press the refresh button to the right of the port entry, it will query all of the serial ports and then populate the ones that exist.  In Windows, it uses the serial.tools.list_ports.comports() command, for linux/cygwin it uses glob.glob("/dev/tty[A-Za-z]*"), and for MacOS it uses glob.glob("/dev/tty.*").  This process is the same that is used in ground control.

I know that it some times takes time to populate the ports on the screen (especially if there is a bluetooth port) and there is no "busy" indicator.

GeroBH, 2019-03-19:
I guess that it's a typo missing the * above.

(Manjaro Linux Python3)
>>> import glob
>>> glob.glob("/dev/tty[A-Za-z]")
[]
>>> glob.glob("/dev/tty[A-Za-z]*")
['/dev/ttyACM0', '/dev/ttyS3', '/dev/ttyS2', '/dev/ttyS1', '/dev/ttyS0']
>>>

madgrizzle, 2019-03-19:
The * is actually there.. looks like it interpreted it as markup code.

`In Windows, it uses the serial.tools.list_ports.comports() command, for linux/cygwin it uses glob.glob("/dev/tty[A-Za-z]*"), and for MacOS it uses glob.glob("/dev/tty.*").  This process is the same that is used in ground control.`

GeroBH, 2019-03-19:
Could it be that:
This was because the arduino was not connected and there for no /ttyACM0 was present?
The S0-3 ports are not given to me as options, so i guess the are filtered out.
That could explain a blank field.

madgrizzle, 2019-03-19:
Weren't you connected to the arduino?  Once you imported gc.ini, it opened the connection so...

There's nothing in the code that would filter responses.  When you go to port setting and hit the refresh button to the right, what shows up (give it a few seconds)?  The software does not auto-populate the list when you bring up settings (it just populates the current port)... you have to hit the refresh button to the right to update the available ports.

GeroBH, 2019-03-19:
Sorry, just edited the post above

GeroBH, 2019-03-19:
It is blank i swear.
[image]
I can bring that back any time.

GeroBH, 2019-03-19:
One more info.
Once the .ini is imported, the port is offered even with a disconnected Arduino.
There is no ttyACM0 on my system now and it is set and offered by WebC.
The refresh button should delete it, right?
The list/field does not seem to be filled with actual ports, but written by a preset.

madgrizzle, 2019-03-19:
When you press the refresh button on the right, do you see "at Update Ports" in the console?

GeroBH, 2019-03-19:
No, just the loop:
connecting
/dev/ttyACM0 is unavailable or in use

GeroBH, 2019-03-19:
Browser??? Let me try.

madgrizzle, 2019-03-19:
Yeah, look to see if there is a javascript error message.

GeroBH, 2019-03-19:
ReferenceError: event is not defined[Learn More] 127.0.0.1:5000:36:9

    <anonymous> http://127.0.0.1:5000/#:36 dispatch2jQuery

madgrizzle, 2019-03-19:
What browser do you use?  I don't use "dispatch2jQuery" in my code so it could just be a module I use (bootstrap, etc.)
When you press the refresh button, what shows up in the browser debugger window?  Any messages?

madgrizzle, 2019-03-19:
I deleted .WebControl file, started webcontrol up on my Windows 10 computer, went to settings and then hit the refresh button to the right of the port field and after 3-4 seconds (because I have bluetooth on my computer) it populated the available ports..  the "actions=updatePorts" is the request being sent to webcontrol from the browser asking for the port list to be refreshed.

[image]

GeroBH, 2019-03-19:
Firefox Quantum 65.0.2 (64-bit)
[image]

GeroBH, 2019-03-19:
The red line repeats with every click

GeroBH, 2019-03-19:
Firefox Quantum 66
[image]
Solved! Can be closed :-)

GeroBH, 2019-03-19:
Believe it or not, Manjaro just offered me an update for firefox and i thought, why not...
And now all is as expected.

madgrizzle, 2019-03-19:
Weird.. I just updated firefox and I get the same error... but its windows 10 firefox.

The problem is related to that one button, I believe, and likely related to how events are managed when a button is part of a form.  I vaguely recall having to hack a solution to make it work and maybe that solution doesn't work across all browsers.  I think the importing of gc.ini is actually unrelated to the problem, other than being a way to populate that field.  Thanks for helping me track down where the problem is.

madgrizzle, 2019-03-19:
Ok, apparently I needed to update the update..  No more error messages and it runs like Chrome.  I'll post back to the forum thread and see if the problem others had was browser related as well.

GeroBH, 2019-03-19:
Well i was to early, if i delete .ini and edit the other file, I do get all ports shown now, but it does not connect if i choose the correct one.
Back a step... But at least we got a bit closer

GeroBH, 2019-03-19:
connecting
 is unavailable or in use
connecting
 is unavailable or in use
connected
699be3e717ad4419b5cc08336eb38040
Attempting to re-establish connection to controller
starting UI
connecting
 is unavailable or in use
I'm connected!
connecting
 is unavailable or in use
connecting
 is unavailable or in use
This is what happens when i try, it connects and looses.

GeroBH, 2019-03-19:
I'm done for today. Leave me a 'to test' list for the tomorrow ;-)

madgrizzle, 2019-03-19:
you are selecting ttyACM0 from the Settings->Maslow Settings and hitting submit?

madgrizzle, 2019-03-19:
No problem.. you are still trying to connect to a "blank" port.   Maybe the submit button isn't working?

GeroBH, 2019-03-19:
Am i missing a bottom part of a window? Where is submit?
Yes, that was it. Scrolling inside the settings window does not bring that button up. Scrolling outside of the settings window does... I really shod go to bed. Sorry.

madgrizzle, 2019-03-19:
On a "large" screen, the button is visible but it looks like the resolution of your browser window is "low".  This is where my inexperience with web-based software really shows.  Trying to make a responsive display is actually quite difficult.  I'll create an issue for that.. the intent was to make the modal fit on the screen with the submit and close buttons on the bottom and then scroll the settings within that modal.

madgrizzle, 2019-04-18:
@GeroBH, safe to close this as the problem seems to be that other UI related screen size issue?  Let me know otherwise.

GeroBH, 2019-04-18:
Safe to close.

closed 2019-04-18 by madgrizzle