WebControlCNC/WebControl #19: Add GPIOs for functionality

madgrizzle, 2019-04-17:
Make a new settings page to configure GPIOs on the RPi for the following functions:

1. HIGH/LOW on WebMCP running/not running
2. HIGH/LOW on WebControl running/not running
3. digitalRead to initiate soft shutdown
4. digitalRead to send ‘stop’ message to controller
5. digitalRead to send ‘pause’ message to controller
6. digitalRead to send ‘home’ message to controller

thejobbitt, 2019-10-02:
This is a super interesting addition, did you have a controller design in mind? Also I added a 2 line character display to output the network ssid and ip address of the machine. It would be awesome to include support for this.

madgrizzle, 2019-10-02:
People asked for the gpio capability so input it in.  If you have code that would output to a 2line display, I can look at incorporating it.

thejobbitt, 2019-10-02:
Display Code
 [I2C_LCD_driver](https://gist.github.com/vay3t/8b0577acfdb27a78101ed16dd78ecba1)
```
#!/usr/bin/python
#includes
import time
import I2C_LCD_driver
import socket
import fcntl
import struct
import subprocess
import psutil

#objects
display = I2C_LCD_driver.lcd()

#main

def system_status(type):
	if type == 'CPU':
		CPU_percent = str(psutil.cpu_percent(interval=5))
		return CPU_percent
	if type == 'DISK':
		DISK_percent = str(psutil.disk_usage('/')[3])
		return DISK_percent
	if type == 'MEM':
		MEM_percent = str(psutil.virtual_memory()[2])
		return MEM_percent

def get_ip_address(ifname):
   		s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    		return socket.inet_ntoa(fcntl.ioctl(
        		s.fileno(),
        		0x8915,
        		struct.pack('256s', ifname[:15])
    		)[20:24])

def get_ssid():
	proc = subprocess.Popen(['/sbin/iwgetid', '-r'], stdout=subprocess.PIPE, stderr=subprocess.PIPE)
	(ssid_name, dummy) = proc.communicate()
	return ssid_name.rstrip()

def bounce(amount):
	for x in xrange(0,amount):
		for x in xrange(0,16):
			display.lcd_clear()
			display.lcd_display_string("*", 1, x)
			display.lcd_display_string("*", 2, 15-x)
			time.sleep(0.03)
			pass
		for x in xrange(0,16):
			display.lcd_clear()
			display.lcd_display_string("*", 2, x)
			display.lcd_display_string("*", 1, 15-x)
			time.sleep(0.03)
			pass
		pass
	display.lcd_clear()

def startup():		
	for x in xrange(0,16):
		display.lcd_display_string("*", 1, x)
		display.lcd_display_string("*", 2, 15-x)
		time.sleep(0.05)
		pass
	display.lcd_display_string("***Maslow CNC***", 1)
	time.sleep(5)

def clear_delay(wait, clear):
	time.sleep(wait)
	if clear == True:
		display.lcd_clear()

bounce(2)
startup()

while True:
	display.lcd_display_string("***Maslow CNC***", 1)
	display.lcd_display_string("%s %s" %(time.strftime("%H:%M"),time.strftime("%m/%d/%Y")), 2)
	clear_delay(5, True)
	bounce(1)
	display.lcd_display_string("SSID %s" %get_ssid(), 1)
	display.lcd_display_string("IP %s" %get_ip_address('wlan0'),2)
	temp_cpu = system_status('CPU')
	bounce(1)
	display.lcd_display_string("CPU USE    %s%s" %(temp_cpu,"%") ,1)
	display.lcd_display_string("DISK USE   %s%s" %(system_status('DISK'),"%") ,2)
	clear_delay(5, True)
	bounce(1)
```

I could put together some build notes. Are you planning on adding a wiki?