WebControlCNC/WebControl #21: Feature Request: Ability to send arbitrary gcode

madgrizzle, 2019-04-18:
Add a screen that allows users to enter gcode and send it, line by line, sequentially.

davidelang, 2019-05-08:
start off with the very simple ability to do something like:

webcontrol/gcode?g0 X0 Y0 G1 Z-5

etc

madgrizzle, 2019-05-12:
I've added a web page to type in and send gcode.

closed 2019-09-14 by madgrizzle

davidelang, 2019-12-15:
this is not working on 0.918

blurfl, 2019-12-15:
Pasting 

```
g0 X0 Y0 G1 Z-5
```
into the 'Send Custom Gcode' form works as expected in 0.918 on a pi-singledirectory installation.

davidelang, 2019-12-15:
10.0.10.50:5000 gets me to webcontrol 10.0.100.50:5000/gcode gets me an 'unknown url' error

madgrizzle, 2019-12-15:
webcontrol uses websockets for the most part so the /gcode isn't something that's routed.  By selecting GCode from the menu and pressing 'Send Custom GCode' it will request the page and should show it in a modal screen.

davidelang, 2019-12-15:
ok, I missed that, from the comment I thought it was just change the url and I would get a form

madgrizzle, 2019-12-15:
Yeah, I see how it could be read that way.

davidelang, 2019-12-15:
having the ability to do something like I was reading it would make it really 
easy to hook in other controls.

David Lang

On Sat, 14 Dec 2019, madgrizzle wrote:

> Yeah, I see how it could be read that way.
ëë>
>