WebControlCNC/WebControl #22: add ability to handle small values

davidelang, 2019-05-08:
if you have a value like 0.000005 it will become 5e-6 and the firmware will choke on it.

this is one of the things holey calibration added, look there for example code

madgrizzle, 2019-05-08:
I ran into this problem.. I'll see he he solved the isclose() problem.

madgrizzle, 2019-05-08:
I ran into this problem.. I'll see how he solved the isclose() problem.

madgrizzle, 2019-05-08:
Looks like there was no change.  I guess the small values in holey calibration aren't small enough to run into the problems I did with optical calibration where the curve fit program came up with coefficients on order of 10e-11... Which is too small to transmit without using scientific notation.  I'll try to revert my the changes in time being.

madgrizzle, 2019-05-08:
I didn't mean to close issue..

davidelang, 2019-05-08:
here it is, bot inline and as an attachment.

David Lang

[Computed Settings]
distperrot = 63.5
kpposmain = 1300
kiposmain = 0
kdposmain = 34
propweightmain = 1
kpposz = 1300.0
kiposz = 0
kdposz = 34
propweightz = 1
kpvmain = 5
kivmain = 0
kdvmain = 0.28
kpvz = 5
kivz = 0
kdvz = 0.28
chainoversprocketcomputed = 2
fpwmcomputed = 3.0
kinematicstypecomputed = 2

[Maslow Settings]
bedheight = 1219.2
openfile = 
motoroffsety = 426.986
sledwidth = 310
zaxissafeheight = 2
bufferon = 0
bedwidth = 2438.4
comport = /dev/ttyACM0
macro1_title = chains to 1495
sledheight = 139
macro2 = 
macro1 = G21 G40 G90 G0 Z3 G0 X0.28 Y526.88 G0 Z0
zaxis = 1.0
sledcg = 79
macro2_title = Macro 2
colorscheme = Light
zdistperrot = 3.17
motorspacingx = 2977.57
sledweight = 98.71

[Advanced Settings]
rotationradius = 138.500
chainextendlength = 1495.0
kiv = 0
gearteeth = 10
spindleautomate = None
kdvz = 0.28
chainsagcorrection = 0.0
chainpitch = 6.35
chainlength = 3360
propweightz = 1
zencodersteps = 7560.0
maxfeedrate = 500.0
enablepospidvalues = 0
rightchaintolerance = -0.492228
enablevpidvalues = 0
maxtouchprobeplungedistance = 0.0
kpvz = 5
kdposz = 34
chainoversprocket = Bottom
kpposz = 1300
fpwm = 490Hz
kdpos = 34
encodersteps = 8113.73
truncate = 1
homex = 0.0
homey = 0.0
kinematicstype = Triangular
kipos = 0
kdv = 0.28
kiposz = 0
digits = 5
kppos = 1300
kpv = 5
propweight = 1
positionerrorlimit = 2000.0
kivz = 0
leftchaintolerance = -0.457313
chainelasticity = .0000081

[Ground Control Settings]
zoomout = pagedown
centercanvasonresize = 0
viewscale = .45
zoomin = pageup
validextensions = .nc, .ngc, .text, .gcode

[Background Settings]
backgroundfile = 
manualreg = []

[Computed Settings]
distperrot = 63.5
kpposmain = 1300
kiposmain = 0
kdposmain = 34
propweightmain = 1
kpposz = 1300.0
kiposz = 0
kdposz = 34
propweightz = 1
kpvmain = 5
kivmain = 0
kdvmain = 0.28
kpvz = 5
kivz = 0
kdvz = 0.28
chainoversprocketcomputed = 2
fpwmcomputed = 3.0
kinematicstypecomputed = 2

[Maslow Settings]
bedheight = 1219.2
openfile = 
motoroffsety = 426.986
sledwidth = 310
zaxissafeheight = 2
bufferon = 0
bedwidth = 2438.4
comport = /dev/ttyACM0
macro1_title = chains to 1495
sledheight = 139
macro2 = 
macro1 = G21 G40 G90 G0 Z3 G0 X0.28 Y526.88 G0 Z0
zaxis = 1.0
sledcg = 79
macro2_title = Macro 2
colorscheme = Light
zdistperrot = 3.17
motorspacingx = 2977.57
sledweight = 98.71

[Advanced Settings]
rotationradius = 138.500
chainextendlength = 1495.0
kiv = 0
gearteeth = 10
spindleautomate = None
kdvz = 0.28
chainsagcorrection = 0.0
chainpitch = 6.35
chainlength = 3360
propweightz = 1
zencodersteps = 7560.0
maxfeedrate = 500.0
enablepospidvalues = 0
rightchaintolerance = -0.492228
enablevpidvalues = 0
maxtouchprobeplungedistance = 0.0
kpvz = 5
kdposz = 34
chainoversprocket = Bottom
kpposz = 1300
fpwm = 490Hz
kdpos = 34
encodersteps = 8113.73
truncate = 1
homex = 0.0
homey = 0.0
kinematicstype = Triangular
kipos = 0
kdv = 0.28
kiposz = 0
digits = 5
kppos = 1300
kpv = 5
propweight = 1
positionerrorlimit = 2000.0
kivz = 0
leftchaintolerance = -0.457313
chainelasticity = .0000081

[Ground Control Settings]
zoomout = pagedown
centercanvasonresize = 0
viewscale = .45
zoomin = pageup
validextensions = .nc, .ngc, .text, .gcode

[Background Settings]
backgroundfile = 
manualreg = []

madgrizzle, 2019-05-11:
> when I uploaded the groundcontrol.ini from holey triangulation, I get a lot of errors from webcontrol about invalid numbers. In my case, the chain stretch is small enough to result in scientific notation. David Lang

So, what I'm not understanding is where you are seeing these errors?  I see chainelasticity = .0000081 but when I do an import into webcontrol, I don't see those errors.. primarily, I think, because webcontrol ignores chainelasticity (since it's not a setting yet)... Is it that these errors occur upon syncing with the controller?

davidelang, 2019-05-11:
I set the values in the firmware using the holey GroundControl and then when I 
connect using WebControl it throws a lot of errors.

David Lang

davidelang, 2019-05-11:
Also, this was running the holey firmeware as well.

On Sat, 11 May 2019, David Lang wrote:

> I set the values in the firmware using the holey GroundControl and then when 
> I connect using WebControl it throws a lot of errors.
>
> David Lang
>

madgrizzle, 2019-05-11:
Well, there's a number of things that it could be, but most likely the problem (and certainly is a problem) is that holey calibration adds some custom firmware values ($45=chainelasticity) that conflict with my optical calibration($45=xyErrorArray).