WebControlCNC/WebControl #24: Macro Buttons don't read Macro1_title or Macro2_title from groundcontrol.ini

wkellyoconnor, 2019-06-02:
I use Macro 1 and Macro 2 to start and stop the spindle. In groundcontrol.ini I have this: macro1_title = Router On
sledheight = 139
macro2 = M05
macro1 = M03
zaxis = 1
sledcg = 79
macro2_title = Router Off
But in WebControl both buttons still show: Macro 1 and Macro 2 but when pressed, they send the proper GCodes and the spindle turns on and off.

madgrizzle, 2019-06-02:
Yeah, this is something from GC that I never really implemented.  I'll look into it, but relabeling buttons and reducing font size automatically to fit that text in I think is going to be a challenge (that's why I never got around to it).  I can implement a straight button text update, but the user may have to limit the number of characters themselves.

BarbourSmith, 2019-06-03:
Yeah, resizing font to fit on a button is remarkably difficult. I really which there was just an option for having the text fit

wkellyoconnor, 2019-06-03:
Ok, I didn't know it was simply not implemented. If you ever do, I'd be happy to test fitting various size text.

gb0101010101, 2020-02-10:
Created [pull request](https://github.com/madgrizzle/WebControl/pull/97) with quick fix to display custom macro title truncated to 6 characters.

madgrizzle, 2020-02-18:
Merged into master.  Closing.

closed 2020-02-18 by madgrizzle