WebControlCNC/WebControl #31: Release 0.901 rendering issues

blurfl, 2019-09-13:
I'm seeing some issues crop up in v0.901 that weren't in the previous version:
- The red cursor doesn't move as the sled moves
- The controller text pane is blank
- The sled position numbers don't populate

I've used Edge and Firefox on Win10, Safari and Firefox on macOS and on iOS, all exhibit the issues. Is there additional data I can provide?

madgrizzle, 2019-09-14:
Hmm. Means you are either not connected or there's a JavaScript error.  Can you look at the browser debugger and see if there are JavaScript errors?

madgrizzle, 2019-09-14:
And does it work fine with v0.900?  I tweaked the firmware checking routines and thought they were working, but maybe something is going amiss there.  Also, let me know what's being reported in the windows console window.. might be something there that can point me in the right direction.  Unfortunately, where I am, I don't have my spare arduino handy

blurfl, 2019-09-14:
It works fine with v0.900. 
v0.901 shows a correct connection, and data appears in the 'Controller Messages' pane created by the console window, but not in any other browser tab, local or remote.
You'll need to look into the JavaScript errors, I'm not useful there.
Here are logs and the console contents for a short session with a simple move.

[logs.zip](https://github.com/madgrizzle/WebControl/files/3612248/logs.zip)

madgrizzle, 2019-09-14:
Thanks for the log files.. for some reason, the firmwares weren't included in the build.  I 'rebuilt' it and they seem to be there now.  I've noticed some flakiness with pyinstaller.. not sure why re-running it would result in different results.  I'll update the release once I can get my laptop reconnected to my home wifi.

madgrizzle, 2019-09-14:
Well, files were there, just that there was a path problem.. long story but its figured out.  I updated the release file.

blurfl, 2019-09-14:
Would the firmware files have an effect on the web page rendering?

madgrizzle, 2019-09-14:
Yes, unfortunately.  but it should be fixed now.  It's a long story related to paths within a pyinstaller package and the thread that processes messages throwing an error and crashing because it tried to convert a "" to a floating point number.

madgrizzle, 2019-09-14:
It did work when I tested it earlier.. but it just happened that I tested it from a folder that had the firmwares in the exact locations that matched what the program was trying to access... so it didn't crash.  Now I know to test it from a different directory,

blurfl, 2019-09-14:
FWIW, I'm interested in building a macOS version. I've got pyinstaller going, and see that there are some directory references that I'll need to change. Are there other things I need to watch for?

blurfl, 2019-09-14:
You've fixed it 👍🏻💯

closed 2019-09-14 by blurfl

madgrizzle, 2019-09-14:
The biggest problem was getting the directory to the dlls correct.  And one dll still doesn't seem to be able to be found, but its for optical calibration so I didn't worry about the error.  I think you'll run into a completely different set of problems considering the differences between macOS and windows.