WebControlCNC/WebControl #34: IP address reported on docker-based system isn't the host's address

madgrizzle, 2019-09-18:
Long story short.. when running in a docker on an RPI, webcontrol reports the docker container's internal ip address, which is not accessible outside the RPI.

To fix, it appears the IP address has to be injected via an environmental variable.  However, WebMCP is used to start the webcontrol docker, so it needs an injection as well.. that means the startup script in the pre-built image needs to inject it.  Lot's of changes to make it work and I'm not sure how to modify everyone's startup scripts.

BarbourSmith, 2019-09-18:
Oooch that sounds like a hassle

madgrizzle, 2019-09-18:
For a vast majority of the cases, the rpi will be headless so you need to know the ip address to get there in the first place.  I'm not sure just how important it is to display it on the screen for this use case.

BarbourSmith, 2019-09-18:
Yeah, it sounds like it might not really be needed to display it on the screen in the first place

madgrizzle, 2019-12-06:
Closing this because its docker related and I don't think I'm going to be able to fix it.

closed 2019-12-06 by madgrizzle