WebControlCNC/WebControl #36: webcontrol-linux pre-release

blurfl, 2019-09-20:
tested on Raspbian Stretch, error:
>./webcontrol: cannot execute binary file: Exec format error

tested on Linux Mint 19.2 in a VM, ran but G-Code, Actions, Settings, Boards actions did not appear. Sub-menus showed but nothing happens when one is selected.

madgrizzle, 2019-10-01:
I don't know why I don't get alerted to new issues.

Anyway, the issue is related to webgl not running on a browser inside the VM, which is not an issue with webcontrol outside the fact that webcontrol uses webgl for the gcode rendering.

closed 2019-10-01 by madgrizzle

BarbourSmith, 2019-10-02:
If you click the "watch" button at the top you should be alerted to new issues through email:

[image]

madgrizzle, 2019-10-02:
I have it set to watch.. maybe I missed it or it ended in spam. /shrug