WebControlCNC/WebControl #37: Z in g code comment causes crash

thejobbitt, 2019-09-20:
I came across this weird issue, where Z in a comment causes WebControl to completely lockup. Ground Control works without issues. And I even tried a new SD card in the raspberry pi with a fresh install. 

G code attached. This code was originally exported from Fusion 360 and it adds a comment about ( ZMIN = something). 

[circle test.txt](https://github.com/madgrizzle/WebControl/files/3636550/circle.test.txt)

madgrizzle, 2019-09-20:
Thanks for finding that and figuring out what the problem is.  That helps immensely.

madgrizzle, 2019-09-20:
It's fixed, but I have to rebuild the docker tonight and push it.

thejobbitt, 2019-09-20:
Awesome, so fast.  I was thinking I was crazy trouble shooting it.

madgrizzle, 2019-09-20:
I used ground control routines to build webcontrol, but ground control parses out all the comments from the file but I keep them in for readability.  However, I had to rewrite certain sections to accommodate the presence of comments and that's one part I missed.  It thought the 'z' in the comment was a move and my routine to send position information to the display choked on it.  Easy enough to fix knowing what the problem was (comments containing 'z').. otherwise, it might have taken a while.  Thanks again.

madgrizzle, 2019-09-21:
Update pushed to docker hub.

thejobbitt, 2019-09-26:
Works good on this end. Thanks for the update.

closed 2019-09-26 by thejobbitt