WebControlCNC/WebControl #38: Firmware update/upgrade error

ted1735, 2019-10-03:
WebControl Maslow Holey firmware upgrade error

Please see link to documented errors.

	* So I am not sure if you wanted to have it ready or if it was ready but I tried to update the firmware to holey so I could calibrate my machine finally with holey calibration.
	* 
Initially it couldn't find the file, so I found it in "etc" folder and copied it to the path it seems it was looking for. That didn't work. So I then added a space as noted below and the ENTIRE correct path to the command line to be passed to avrdude.
	* 
There seems to be an error in the path and maybe just a small typo of a missing space. I am not familiar with the handles/switch commands involved with avrdude or how they are passed through. However just by viewing the python window I could see there was an error and surmise that the "-C" probably needed a space before the path was passed through to avrdude to look for avrdude.conf.
	* 
AND IT WORKED!!!
	* 
Below are screen shots you can follow if you choose or just look at the spacing and path that is passed to avrdude and it will fix it!!

[image]
[image]
[image]
[image]

https://www.evernote.com/shard/s28/client/snv?noteGuid=2680c421-3a14-49ea-8fb6-0261a8761a73&noteKey=633eee4998e63ccc&sn=https%3A%2F%2Fwww.evernote.com%2Fshard%2Fs28%2Fsh%2F2680c421-3a14-49ea-8fb6-0261a8761a73%2F633eee4998e63ccc&title=WebControl%2BMaslow%2BHoley%2Bfirmware%2Bupgrade%2Berror

madgrizzle, 2019-10-03:
Thanks for finding that issue.  I think it might be a difference between Linux and windows versions of avrdude, but regardless, I see some things I need to xhange/include (such as avrdude itself) in the pyinstaller files.

blurfl, 2019-10-03:
Running the 'webcontrol-0.906-rpi-singledirectory' version, the firmware update option seems to be looking for avrdude on a Windows-style path, so it fails. The web page reports 'update complete' though.

```

pi@pi4:~/webcontrol $ ./webcontrol 
Initializing Logger
/home/pi/.WebControl/log.txt
/home/pi/webcontrol
Initializing Configuration
Camera initialized
##
raspbian gnu/linux
##
----
rpi
singledirectory
----
VIDEOIO ERROR: V4L: can't open camera by index 0
/home/pi
Starting Console Queue Processor
/home/pi/.WebControl/webcontrol-*.port
-$$$$$-
/home/pi/webcontrol/main.py
/home/pi/webcontrol
-$$$$$-
opening browser
check for pyrelease
0.906
v0.906
webcontrol-0.906-rpi-singledirectory.tar.gz
https://api.github.com/repos/madgrizzle/WebControl/releases/assets/15259000
{"hostAddress": "127.0.1.1:5000"}
connected
1c09b890a2ee4e4ba5021281ddd00b92
Attempting to re-establish connection to controller
starting UI
connecting

Connected on port /dev/ttyACM0

Sent G20
I'm connected!
Sending Board Data
Sent Board Data
Sending Board Cut Data compressed
Sent Board Cut Data compressed
Sending: B05  
WebControl 1.27
Firmware Version 1.28

Sending: G20  
Sending: $$ 
/home/pi/webcontrol/firmware/maslowcnc/*.hex
portClosed0
connection closed at serialPortThread
/dev/ttyACM0
/home/pi/webcontrol
/home/pi/webcontrol/firmware/maslowcnc/maslowcnc-1.26.hex
"C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avrdude" -Cavr/avrdude.conf -v -patmega2560 -cwiring -P/dev/ttyACM0 -b115200 -D -Uflash:w:/home/pi/webcontrol/firmware/maslowcnc/maslowcnc-1.26.hex:i
sh: 1: C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avrdude: not found
32512
closing modals
{"title": "Actions"}
connecting

Connected on port /dev/ttyACM0

Sent G20
Sending: B05  
WebControl 1.27
Firmware Version 1.28

Sending: G20  
Sending: $$ 
/home/pi/.WebControlwc_diagnostics_20191002-215659.zip
PYINSTALLER
pi@pi4:~/webcontrol $ 

```

madgrizzle, 2019-10-03:
Yeah, I'm going to see if I can package avrdude in the pyinstaller.

madgrizzle, 2019-10-03:
So, here's something to overcome.. The goal is to be able to install webcontrol and be able to update the firmware with just a single install file.  I seem to be able to run avrdude by packaging both avrdude.exe and libusb0.dll into the pyinstaller.  But I worry about the driver for the arduino.  I looked at the driver file details for the arduino connected to my Win10 computer and its:

>C:\WINDOWS\system32\DRIVERS\usbser.sys

This is a Microsoft provided driver and I think it's built-in to Windows 10.. so does anyone know if a different driver is needed?

What about Linux and RPI?  Anyone know enough about this stuff to know?

madgrizzle, 2019-10-03:
I created a new version, but my computer is reporting it as a virus.  Another computer seems fine though.  I noticed the troubled computer hasn't updated in nearly a year.. so going through the long and excruciating process of trying to update.

davidelang, 2019-10-04:
Linux/Pi already has the driver on the system

OS/X and older windows versions may need a driver to communicate, but that's a 
one-time thing, upgrades don't change this.

David Lang

On Thu, 3 Oct 2019, madgrizzle wrote:

> So, here's something to overcome.. The goal is to be able to install webcontrol and be able to update the firmware with just a single install file.  I seem to be able to run avrdude by packaging [...]

madgrizzle, 2019-10-04:
My windows updates keep failing, but at least the virus definition file updated and doesn't report webcontrol as a trojan horse.  Hopefully will have new releases to test out sometime today.

madgrizzle, 2019-10-04:
Please all try again with latest (I deleted previous release).  Win32 builds will be added soon.

blurfl, 2019-10-04:
Firmware update 'Stock' works here on Win10-64 and various Pi's. 💯

I could wish that there was a firmware option for me to burn a local version of firmware... 😁 For now I can keep replacing a file in the singledirectory-firmware folder, but that's harder with the singlefile version.

madgrizzle, 2019-10-04:
In think I could add a local firmware option located using the .WebControl directory.

blurfl, 2019-10-04:
That would be excellent! 👍

madgrizzle, 2019-10-05:
Win32 versions added for testing.

ted1735, 2019-12-03:
**Error occurred when trying to upgrade firmware to holey for calibration.**

**### After trying to build the ftdi and usblib files, and finding dependency issues with udev and udev-dev, i then installed avrdude at the command line and it added the proper ftdilib!!!**

/home/ted/webcontrol/tools/avrdude: error while loading shared libraries: libftdi.so.1: cannot open shared object file: No such file or directory
connecting

/home/ted
Starting Console Queue Processor
/home/ted/.WebControl/webcontrol-*.port
-$$$$$-
/home/ted/webcontrol/main.py
/home/ted/webcontrol
-$$$$$-
opening browser
check for pyrelease
/usr/bin/exo-open: /home/ted/webcontrol/libz.so.1: version `ZLIB_1.2.9' not found (required by /usr/lib/x86_64-linux-gnu/libpng16.so.16)
/usr/bin/exo-open: /home/ted/webcontrol/libz.so.1: version `ZLIB_1.2.9' not found (required by /usr/lib/x86_64-linux-gnu/libpng16.so.16)
0.917
None
XPCOMGlueLoad error for file /usr/lib/firefox/libmozgtk.so:
/home/ted/webcontrol/libz.so.1: version `ZLIB_1.2.9' not found (required by /usr/lib/x86_64-linux-gnu/libpng16.so.16)
Couldn't load XPCOM.
XPCOMGlueLoad error for file /usr/lib/firefox/libmozgtk.so:
/home/ted/webcontrol/libz.so.1: version `ZLIB_1.2.9' not found (required by /usr/lib/x86_64-linux-gnu/libpng16.so.16)
Couldn't load XPCOM.
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: iceweasel: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: seamonkey: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: mozilla: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: epiphany: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: konqueror: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: chromium: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: chromium-browser: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: google-chrome: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: www-browser: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: links2: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: elinks: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: links: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: lynx: not found
/usr/bin/xdg-open: 851: /usr/bin/xdg-open: w3m: not found
xdg-open: no method available for opening 'http://localhost:5000'
connecting

Connected on port /dev/ttyACM0

{"hostAddress": "127.0.1.1:5000"}
connected
537a01b235594902be170acb5a157fc9
starting UI
I'm connected!
Sending Board Data
Sent Board Data
Sending Board Cut Data compressed
Sent Board Cut Data compressed
537a01b235594902be170acb5a157fc9
uncompressed:1349064
compressed3D:171319
Sending Gcode compressed
Sent Gcode compressed
uncompressed:2
compressed3D:22
Sending Gcode compressed
Sent Gcode compressed
537a01b235594902be170acb5a157fc9
processing request to close
connection closed at serialPortThread
/home/ted/webcontrol/tools/avrdude: error while loading shared libraries: libftdi.so.1: cannot open shared object file: No such file or directory
connecting

Connected on port /dev/ttyACM0
[... 199 lines cut ...]
Sending: $2=2996.97
Sending: $3=413.4
Sending: $5=152.4
Sending: $7=2
Sending: $8=131.2
Sending: $11=1650
Sending: $15=750
Sending: $29=1300.00
Sending: $37=102.498904
Sending: $38=2.00000
Sending: $45=0.00000516850
Sending: $46=97.9000
Sending: $2=2996.97
Sending: $3=413.4
Sending: $5=152.4
Sending: $7=2
Sending: $8=131.2
Sending: $11=1650
Sending: $15=750
Sending: $29=1300.00
Sending: $37=102.498904
Sending: $38=2.00000
Sending: $45=0.00000516850
Sending: $46=97.9000
537a01b235594902be170acb5a157fc9

madgrizzle, 2019-12-03:
With which operating system didn't this occur?

ted1735, 2019-12-04:
Linux zorin latest

madgrizzle, 2019-12-04:
Could this possibily Zorin thing?  I don't have a linux machine with an arduino connected to it (I just run linux on a VM and don't have any hardware connected to the host box) so it's hard for me to test.  But, I did search my linux VM (ubuntu 16.04) and I was able to find ftdi related files (for example, libtdi.so.1 that I saw was marked as missing in your log).. So are these files not included in Zorin's distro or is there something I else going on?

blurfl, 2019-12-05:
>Could this possibily Zorin thing?

I see the same thing in Mint19 using the singleDirectory-0.917. Replacing the tools/avrdude executable with a newer version (from Arduino 1.8.10 in this case) solves the issue and allows the webcontrol/Actions/firmwareUpgrade process to complete properly.

madgrizzle, 2019-12-05:
So is it an avrdude version issue only or is there also a driver issue for the usb?

blurfl, 2019-12-05:
I would guess that it's an avrdude version issue. The newer version reports "Version 6.3-20190619". No other change was needed to restore operation beyond replacing that one executable file. I think that version is built using the scripts [here](https://github.com/arduino/avrdude-build-script).

madgrizzle, 2019-12-05:
So this new version contains all the missing files and @ted1735 wouldn't have needed to build the ftdi and usblib files?

blurfl, 2019-12-05:
That's how it looks to me. Just replicated the situation on a Fedora 30 VM; singleDirectory-0.917 firmware update failed and replacing the tools/avrdude with the newer version solved it.

madgrizzle, 2019-12-05:
great.. will try to figure out how to get a copy of it.  apt-get only seems to get me 6.25

madgrizzle, 2019-12-05:
ok, got it built.  Will try to build releases tonight and post a new version.

ted1735, 2019-12-06:
sound great! Its always hard when you feel like you are the only one having an issue, however glad it was able to be replicated and discovered!! If you need any more info LMK!! I run my apps off an old Intel dual core laptop, and a pi 3b+ so if you need something in particular tested. I am happy to help. Webcontrol is such an amazing and innovative program and has made using Maslow so much easier and more efficient!!

blurfl, 2019-12-06:
the singleDirectory v0.918 seems to have done the trick on the Mint19 and Fedora31 VMs. 💯👍

madgrizzle, 2019-12-06:
> I am happy to help.

Thanks for testing this out and the research in determining the problem.  It saves me countless, countless hours.. especially for things I'm not very familiar with.