WebControlCNC/WebControl #40: Gcode line containing comments causes loss of connection

blurfl, 2019-10-05:
webcontrol correctly handles separate comment lines enclosed in parentheses, but if a line with a valid gcode has a comment at the end, the serial connection is lost. The following snippet will demonstrate the issue, causing a disconnect when run. Remove the comment on the 'G90' line and the snippet will run correctly. 
```
( G-Code Modified by G-Code Ripper )  
( by Scorch - 2013-2019 www.scorchworks.com )  
G90 ( this comment will demonstrate the issue ) 
G17    
G21    
(G-Code Preamble)  
(---------------------------------------------------------)  
G0 X0.0 Y0.0 Z0.0000    
G3 X-0.0178 Y0.0178 Z-12.7000 I17.9603 J17.9603 F2540.00    
G0 X0.0 Y0.0    
G2 X-0.0178 Y0.0178 Z0.0000 I-17.9603 J-17.9603    
```

madgrizzle, 2019-10-05:
Hmmm. I assume this doesn't occur in gc because gc strips the comments.  So maybe the firmware chokes on it.  Will work on stripping it out.  Is there any other form of comment delineators other than ( )?

blurfl, 2019-10-05:
GC dows its stuff in gcodecanvas.py, lines 156, 157:
```
            filtersparsed = re.sub(r'\(([^)]*)\)','\n',rawfilters) #replace mach3 style gcode comments with newline
            filtersparsed = re.sub(r';([^\n]*)\n','\n',filtersparsed) #replace standard ; initiated gcode comments with newline

```

blurfl, 2019-10-05:
[Here's](http://linuxcnc.org/docs/html/gcode/overview.html#gcode:comments) the word from linuxcnc.org

madgrizzle, 2019-10-05:
I certainly can filter the line before sending it.. I just dont want to filter upon loading as gc does.

davidelang, 2019-10-06:
it would be better to fix the firmware to ignore comments rather than strip them 
in GC, for compatibility with other g-code senders

stripping them in GC is an optimization to save data sent over the serial port, 
but there are typically very few comments, so it's not a significant one.

David Lang

blurfl, 2019-10-06:
Stripping the comments before sending them helps with the 60-character line limit in the firmware.

 The Maslow firmware has most of grbl's comment-handling code already implemented. There's a one-line patch needed to make it handle comments correctly that takes care of the example above, I'll submit that to the Maslow Firmware repo.

There is a slightly different twist here in WebControl - if a comment contains one of the characters {xyz} then WC chokes. Sending this gcode line with XY in the comment:
```
G17 (set active plane to XY)  
```
results in this in the terminal:
```
Sending: G17 (set active plane to XY)  
Exception in thread Thread-4:
Traceback (most recent call last):
  File "threading.py", line 914, in _bootstrap_inner
  File "site-packages/gevent/threading.py", line 177, in run
  File "threading.py", line 862, in run
  File "Connection/serialPortThread.py", line 260, in getmessage
  File "Connection/serialPortThread.py", line 109, in sendNextLine
  File "Actions/actions.py", line 1086, in sendGCodePositionUpdate
  File "Actions/actions.py", line 1246, in findPositionAt
ValueError: could not convert string to float: 
```

On a side note, I'm sure enjoying the gcode editor and firmware uploader in WC! 🎉💯 They make this kind of investigation much easier 👍

madgrizzle, 2019-10-06:
Thanks for finding that. I'll get that updated.

madgrizzle, 2019-10-06:
Does ground control handle the comments that come after a valid gcode?

blurfl, 2019-10-06:
Yes, it strips them correctly. 
Is it the case that actions.py:: findPositionAt() is looking for XYZ parameters in the gcode line, but doesn't recognize comments? If so, then processGCode() might get fooled by things in comments as well.

madgrizzle, 2019-10-06:
Thanks.. I went ahead and created a test routine to make sure i understand what the re commands are doing (that command is voodoo in my opinion).   Unfortunately, I don't have a fake servo controller where I'm at.  I might need to order yet another mega to do test work on.

madgrizzle, 2019-10-06:
@blurfl, can you test a single release before I build and update everything?  If so, let me know which variant (x86, x64, linux, rpi.. single file, single directory, etc.) is easiest for you to test.  Without a spare arduino, I'm limited on what I can test with this issue.

blurfl, 2019-10-06:
I’d be happy to! rpi singledir would be easiest.

madgrizzle, 2019-10-06:
release released.

blurfl, 2019-10-06:
Installed the release but recieve this error on ruinning:
```
pi@pi4:~/webcontrol-test2 $ ./webcontrol 
Initializing Logger
/home/pi/.WebControl/log.txt
/home/pi/webcontrol-test2
Initializing Configuration
Camera initialized
##
raspbian gnu/linux
##
----
rpi
singledirectory
----
VIDEOIO ERROR: V4L: can't open camera by index 0
/home/pi
Starting Console Queue Processor
/home/pi/.WebControl/webcontrol-*.port
-$$$$$-
/home/pi/webcontrol-test2/main.py
/home/pi/webcontrol-test2
-$$$$$-
opening browser
check for pyrelease
0.907
v0.907
webcontrol-0.907-rpi-singledirectory.tar.gz
https://api.github.com/repos/madgrizzle/WebControl/releases/assets/15322078
{"hostAddress": "127.0.1.1:5000"}
connected
37305b99feb7431bb20c068a4e1447e6
Attempting to re-establish connection to controller
starting UI
connecting

Connected on port /dev/ttyACM0

Sent G21
I'm connected!
Sending Board Data
Sent Board Data
Sending Board Cut Data compressed
Sent Board Cut Data compressed
Exception in thread Thread-4:
Traceback (most recent call last):
  File "threading.py", line 914, in _bootstrap_inner
  File "site-packages/gevent/threading.py", line 177, in run
  File "threading.py", line 862, in run
  File "Connection/serialPortThread.py", line 237, in getmessage
  File "Connection/serialPortThread.py", line 38, in _write
NameError: name 're' is not defined
```
Side note: the 'Update WebControl: ' button worked a treat 😁

madgrizzle, 2019-10-06:
Glad you tested the update feature.  I'm uploading a new release.. will be there in a minute.  unfortunately, the silly bug couldn't be found until a connection to a controller was established.. so I ordered mega from amazon to put on my build machine.

blurfl, 2019-10-06:
Do you need a board to go on that Mega? I've got spares, glad to send one.

blurfl, 2019-10-06:
That release fixed findPositionAt(), but processGCode() needs the same attention. This line demonstrates it:
```
G21 ( G20 ) 
```
When the gcode loads, the G20 changes the screen to "Inches" mode.

madgrizzle, 2019-10-06:
Thanks for the offer, but it won't be connected to motors.

Building a new version with filters in processGCode() as well.

blurfl, 2019-10-06:
I forgot that a boardless Mega shows up as a v1.2 Maslow 😄

madgrizzle, 2019-10-06:
release re-released.

blurfl, 2019-10-06:
We're getting closer.  It seems to be a WC screen thing at this point. Executing this snippet in the _very excellent_ 'Send Custom GCode' tool changes the screen from 'MM' to 'INCHES' and shows the red cursor at XPos:2.00 Ypos:2.00. Looking in the log.txt file, Mega is still in 'MM' mode and correctly shows its position as 2.00,2.00 millimeters.
```
G90
F1000.0   
G21    
G0 X0.0 Y0.0 Z.0.0    
G1 X1.0 Y1.0 ( G20 )  
G1 X2.0 Y2.0    
```

[downloadDiagnostics-6.zip](https://github.com/madgrizzle/WebControl/files/3694841/downloadDiagnostics-6.zip)

madgrizzle, 2019-10-06:
Thanks for putting it through the paces.  I have another release being built that should fix that.  I think I'm 'over-filtering' at this point (filtering after its been filtered) but that's better than 'under-filtering'.

madgrizzle, 2019-10-06:
release re-re-released

blurfl, 2019-10-09:
Sorry to be so long in replying. I've been trying to zero in on better information for you. Here's what I think is the situation at the most recent test release:

- WC findPositionAt() now correctly ignores XYZ in comments as described [in this comment above](https://github.com/madgrizzle/WebControl/issues/40#issuecomment-538715185) 
- WC currently strips comments off gcode lines before sending to the firmware. I'm using patched firmware PR#534 merged in to current master, which correctly handles comments. Stripping comments from the serial stream will be unnecessary with the next release of firmware.
- the screen rendering code which runs after loading, viewing or editing a file incorrectly operates to render the character sequences found within comments instead of ignoring them. Comments that contain G0, G1, G2 or G3 commands draw on screen. If there is a G20 or G21 in a comment, it can affect the units button. If the last G20/G21 command was in a comment, the units on screen might not match the gcode file. The jog arrows will move the sled the appropriate amount but the red screen cursor and Units button will reflect the error.

madgrizzle, 2019-10-09:
Lol..thanks for finding that.. maybe that's why @bar just stripped all comments out upon loading.  I'll look into the rendering parsing routines.

bar, 2019-10-09:
@madgrizzle I did what?

madgrizzle, 2019-10-09:
sorry @bar.. meant @BarbourSmith (he's @bar on the discourse forum for this project)

bar, 2019-10-10:
@BarbourSmith Ber here! they call me Bär and I'm and eng so it was the most obvious choice :')

Thanks for the tip, I learnt about it almost when they released the feature haha, I get lots of unintended attention!

blurfl, 2019-10-14:
v0.909 handles comments correctly, ignoring them when rendering gcode on the screen 💯🎉
In the process of testing that, I had numerous issues with buffer overflow in the firmware. Here's a snippet that reliably causes it:
```
( demonstrate buffer overflow issue)  
( by Scorch - 2013-2019 www.scorchworks.com )  
G90 (SET ABSOLUTE DISTANCE MODE)    
G17 (set active plane to XY)  
G21 (SET UNITS TO MM)    
F1000    
(G-Code Preamble)  
G0 X0.0 Y0.0 Z0.0000    
G3 X-0.0178 Y0.0178 Z-12.7000 I17.9603 J17.9603 F2540.00    
G0 X0.0 Y0.0    
G2 X-0.0178 Y0.0178 Z0.0000 I-17.9603 J-17.9603    
```
The snippet runs properly in GC, maybe because GC:serialPortThread watches for the 'ok' messages and throttles sending lines to avoid the issue.
 Does this belong in a separate issue?

madgrizzle, 2019-10-14:
I ran into buffer overflow issues as well, even if buffering is turned off.  And it's only on code you provided.  If it runs on gc correctly, I'll take another look to try to figure out what's going on.

madgrizzle, 2019-10-14:
Does this issue occur if all comments are stopped out?

blurfl, 2019-10-15:
The code snippet [above](https://github.com/madgrizzle/WebControl/issues/40#issuecomment-541514935) runs correctly in GC and in WC - _sometimes_, other times it causes a buffer overflow. It doesn't seem to be limited to comments, though. I've been using the 'GCode View/Edit' function, and that seems to cause the issue by adding blank lines to this snippet at the end of the edit window.

 Here's a way to make a buffer overflow happen with extra lines added by WC at then end of the code snippet:
  Use 'Clear GCode' then 'View/Edit' and paste in the above snippet, note there are 11 lines in the edit window. 'Submit' and 'Play', see the code runs correctly. Now 'View/Edit' to look at the code, note that the window shows 13 lines, the last two blank. Leave it alone and instead of 'Submit', choose 'Close ' and then 'Play' to see that the code still runs as expected even with the two additional lines. Now 'View/Edit' and 'Submit' to exit the editor, which will add two more blank lines. After several of these 'Edit/Submit' sequences, enough blank lines accumulate at the end of the code to cause the buffer overflow when the gcode is run. The snippet has two very long gcode lines which exacerbate the issue.

madgrizzle, 2019-10-15:
I've noticed the extra lines.. and I think it's adding trailing spaces as well.  I'm not sure if its the editor doing it or the code that I grabbed from ground control to process what's submitted.  But try the new releases as I've done some cleanup on the filtering.

blurfl, 2019-10-15:
I think you've done it! 💯 I haven't been able to trigger a buffer overflow with the v0.910 code, and it seems to handle all forms of inactive comments properly. Well done!

blurfl, 2019-10-15:
I think this issue could be closed 👍🎉

blurfl, 2019-10-15:
This could be a new issue with v0.910 if you want...
In the terminal, I'm getting a repeating error "index error when reading gcode" when starting v0.910 that repeats rapidly until I've opened a gcode file from the menus. (rpi-singledirectory) If no serial connection has been made, the error doesn't appear until the connection is established.

davidelang, 2019-10-15:
this may be old new, but just in case it's not.

the communications to the arduino does not use any flow control, it's strictly a 
'blast it out and hope the other end has enough space' model. This generally 
works as g-code lines tend to be short, but it sounds like you are running into 
cases where there are longer lines.

David Lang

madgrizzle, 2019-10-15:
Do you have 'buffering' enabled in settings?

blurfl, 2019-10-15:
>the communications to the arduino does not use any flow control, it's strictly a
'blast it out and hope the other end has enough space' model. 

 GC has had flow control for more than a year, and v0.910 of WebControl has ironed out the wrinkles in its implementation. 

serialPortThread.py manages flow control, responding to the 'ok' message sent by the firmware on completion of a gcode line and keeping track of the number of characters of buffer space consumed by each line sent and ack'ed. New lines are sent as 'ok' acks indicate space is available in the firmware buffer. You can set the verboseDebug in the firmware file Config.h to 1 for single-line messages with each gcode line handled or 2 to also output ring buffer contents.

davidelang, 2019-10-15:
flow control as in 'send a line wait for an ok' is not the same as flow control 
at the serial level (which can happen on a per byte level, not a per line level)

are you saying we now have software flow control enabled (control-S, control-Q)? 
the arduinos are not wired to provide hardware flow control (cts/rts)

David Lang

On Tue, 15 Oct 2019, Scott Smith wrote:

> 'blast it out and hope the other end has enough space' model.
>
> GC has had flow control for more than a year, and v0.910 of WebControl has ironed out the wrinkles in its implementation. [...]

blurfl, 2019-10-15:
>Do you have 'buffering' enabled in settings?

The index error happens when the buffering is turned on (blue, not gray). With buffering turned off, there are no index errors, but buffer overflows happen.

madgrizzle, 2019-10-16:
Oh, so you still see buffer overflows, but only with buffering turned off.. correct?

madgrizzle, 2019-10-16:
I found the problem with the index errors, but I wonder if it also exists in ground control.  Looking at the code I can't see why it wouldn't since the routines are mostly the same.  Anyway, I put in a fix for it and will rebuild, but please confirm whether or not you are seeing buffer overflows with buffering turned off.  I can't seem to duplicate it anymore.

blurfl, 2019-10-16:
>please confirm whether or not you are seeing buffer overflows with buffering turned off

Sorry I've taken so long to respond.
Were there multiple releases under 'v0.910'? I seemed to have had different versions of v0.910 on different Rpis, one of which (pi-zero-w) could be made to cause buffer overflows with a suitably gnarly gcode file. The other two (pi3B, pi4), send the file without overflow regardless of the state of the switch. All showed 0.91. After re-downloading v0.910 to the pi-zero-w, it now handles the file without overflow as the others do, with the switch in either state.
 GC does not post index errors in either state of the switch. I've noticed that WC doesn't render the gcode file on startup, though it does remember the file chosen in the 'GCode/Open' dialog.

blurfl, 2019-10-17:
v0.911 solves the 'index error' issue. 👍💯