WebControlCNC/WebControl #41: 'Backup WebControl' creates 22-byte .zip file

blurfl, 2019-10-17:
Not sure whether this function is appropriate to test, but it creates and sends a 22-byte .zip file which is unusable.

madgrizzle, 2019-10-17:
Everything is appropriate to test.  Which version did this?

⁣Sent from BlueMail ​

blurfl, 2019-10-17:
It was v0.911, sorry, I should have indicated...

madgrizzle, 2019-10-17:
Rpi, linux ,win, single file or single directory or all of the above?

⁣Sent from BlueMail ​

blurfl, 2019-10-17:
RPi singledir was where I saw the issue.  Just installed Win64 singlefile and it seems to have worked  properly.

blurfl, 2019-10-18:
Just ran WC from the v0.911 source.zip on a mac, and 'Backup...' has correctly sent a zipped file and created the wc....zip file in my home directory. 
 I ran v0.911 singledir on a Mint linux vm and got a 22-byte .zip file. Ran the v0.911 singlefile version on the same machine and got a 22-byte .zip file as well.

blurfl, 2019-10-18:
It looks like the rpi and win singledir versions use the firmware file to send to the Arduino from within that singledirectory, but the singlefile uses .WebControl/firmware.
 Too, the win singledir version offers (sic) 'Upgrade Stock Firmware:singledir/firmware/maslowcnc/maslowcnc' while the rpi singledir offers 'Upgrade Stock Firmware:1.xx' where the version number seems to be read from the .hex filename

madgrizzle, 2019-10-18:
Can you test the backup again?  My 0.911 didn't work either, so I made an innocuous change to see what was going on (print out a directory name) and now it works.  I don't think the change I made fixed anything.. I'm not sure what happened.. maybe a pyinstaller build issue?

madgrizzle, 2019-10-18:
"It looks like the rpi and win singledir versions use the firmware file to send to the Arduino from within that singledirectory, but the singlefile uses .WebControl/firmware."

I just uploaded from Win64 singlefile and it pulled it from the pyinstaller temp directory.

"Too, the win singledir version offers (sic) 'Upgrade Stock Firmware:singledir/firmware/maslowcnc/maslowcnc' while the rpi singledir offers 'Upgrade Stock Firmware:1.xx' where the version number seems to be read from the .hex filename"

I'm not seeing that on my win64 machine..

blurfl, 2019-10-18:
I just updated an Rpi to v0.912 and the backup created a proper .zip file👍 Other testing after coffee is made. ☕️

madgrizzle, 2019-10-18:
Thanks.. weirdness for sure.

blurfl, 2019-10-18:
Updated the win64 singledir to v0.912 and the backup file remains good, the menu text remains odd. The singledir directory is within my Documents directory.
 The win64 singlefile v0.912 backup still works as expected, the menu option still reads 'Upgrade Stock Firmware:1.26'

madgrizzle, 2019-10-18:
How do you run the singledir version in windows?  Via a shortcut?

madgrizzle, 2019-10-18:
Ah.. what is the full path of the directory?  Does the path happen to have a "-" in it?

madgrizzle, 2019-10-18:
nm.. that's the problem.  I split on the '-' and its finding it in the path name.  I'll fix it :)

madgrizzle, 2019-10-18:
Try out 0.913... hopefully fixed (works on mine)

blurfl, 2019-10-18:
Yes, win64 v0.913 singledir correctly handles a dash in it's directory. 👍🏻

madgrizzle, 2019-10-18:
I'm going to close this thread, but please continue and make new issues as they appear :)

closed 2019-10-18 by madgrizzle