WebControlCNC/WebControl #42: Buffer Overflow, turns off spindle but does not stop movement.

thejobbitt, 2019-10-25:
I'm getting a buffer over flow that kills the spindle signal but then continues to move as if nothing is wrong. The code is generated from Fusion 360, it happens in the first min of the code. The gcode was too long to add but I can forward it. I will add the console output as soon as I am able to rerun it and grab it. Is there a log of all the commands sent?

I am able to run the program fine if I plug the spindle in to a non controlled outlet and turn it on manually.

madgrizzle, 2019-10-25:
Email the code to madgrizzlemaslow at gmail.com. Don't worry about sending lots yet, I can normally duplicate problems from the gcode.  What version of webcontrol are you using (docker or pyinstaller release)?

thejobbitt, 2019-10-25:
I'm using webcontrol on Raspberry Pi 3b, 1.27 release with 1.28 stock firmware.

madgrizzle, 2019-10-25:
Do you have gcode buffering enabled in settings?  I'll look at it tomorrow.  Also, are you using the docker version or one of the pyinstaller releases?

⁣Sent from BlueMail ​

thejobbitt, 2019-10-25:
Buffer gcode is off, and I used the pre-built RPi image. Which looks like is docker?

madgrizzle, 2019-10-25:
Yes, the prebuilt is a docker image.  I'll look into it today.

madgrizzle, 2019-10-25:
I'm going to do some tweaking to the code, but in the meantime, just delete any comments from the gcode file and it should run.  I was able to duplicate the buffer overflow on my end and when I deleted the comments, it didn't overflow.  Comments in that file are bracketed by '(' and ')'.. lines 1, 2 and 6.

madgrizzle, 2019-10-28:
I updated the docker image.  Update using WebMCP and let me know if the problem is resolved (resolved on my end during testing).

thejobbitt, 2019-10-31:
Issue appears to be fixed. Thanks for the quick update.

closed 2019-10-31 by thejobbitt