WebControlCNC/WebControl #46: webcontrol-0.914-rpi-singledirectory run error

blurfl, 2019-11-11:
On an Rpi4, webcontrol installed from 'webcontrol-0.914-rpi-singledirectory.tar.gz' errs out when starting up:

```
pi@pi4:~/webcontrol-v914 $ ./webcontrol 
Traceback (most recent call last):
  File "main.py", line 2, in <module>
  File "/home/pi/.virtualenvs/webcontrol/lib/python3.5/site-packages/PyInstaller/loader/pyimod03_importers.py", line 627, in exec_module
  File "app.py", line 4, in <module>
ImportError: No module named 'flask_misaka'
[1193] Failed to execute script main
```
The earlier version v-911 works correctly on that Rpi.

madgrizzle, 2019-11-11:
I forgot to update all my build vms with the new module requirement.  Will rebuild.

⁣Sent from BlueMail ​

madgrizzle, 2019-11-12:
I got it rebuilt with flask-miska.

blurfl, 2019-11-12:
That did the trick 👍🏻💯

closed 2019-11-12 by blurfl

madgrizzle, 2019-11-12:
Great!