WebControlCNC/WebControl #48: WebControl app for MacOS

blurfl, 2019-11-19:
I've come up with a process to build an app version of WebControl for MacOS. It uses a modified version of main.spec for pyinstaller.

The main issue to overcome was that the app built by pyinstaller for MacOS doesn't open a terminal window the way the Windows version does. That seems to be unavoidable. The app that pyinstaller builds runs properly in the background, but the user can lose track of whether it is running and fire it up a second time and that can leave port 5000 locked, or leave the app in an unresponsive state where 'force quit' or kill don't help. At that point, a power-switch reset is the only solution I found... 

 The workaround I came up with uses pyinstaller to package a version of the webcontrol backgrounder executable then uses platypus to create a wrapper around that. The wrapper checks for any instance of the backgrounder already running and kills that before re-running it. It also puts up an alert window of its own and suggests that the user keep that window open to mark that WebControl is running. The fact that the window is open also puts an icon onto the Dock, a further reminder. If that window is closed, the backgrounder remains running and WebControl continues to work, and re-running the app avoids the issue above.

 I haven't seen an icon for WebControl, so I the Maslow 'M' turned upside down for a 'W':

[image]

BarbourSmith, 2019-11-19:
> I haven't seen an icon for WebControl, so I the Maslow 'M' turned upside down for a 'W':

That looks fantastic! I am not a Mac person, but the theory there makes sense to me

madgrizzle, 2019-11-19:
I created a favicon.ico for the web browser by doing the same (inverting the image) but I colored it blue.  The file, however, is 16x16 (size of a web icon).  I didn't create an "icon" for the program though but makes sense to do.. I'll use yours and turn it blue to match the favicon.

As for the MacOS pyinstaller, how does it work with ground control?  Does gc open up a terminal window?

blurfl, 2019-11-19:
>I created a favicon.ico for the web browser

Oops! I never noticed the favicon. I'll go blue!  
[image]

>As for the MacOS pyinstaller, how does it work with ground control?

GC uses buildozer to build the app; buildozer doesn't have the same issue (or rather,  a different set of issues).

madgrizzle, 2019-11-19:
I'll be pushing an update tonight that includes a multi-image icon and updated .spec files to use it.  I will also be rename main.exe to webcontrol.exe (I've been doing it after the fact in my build scripts.. silly to not just name it correctly in the first place).

blurfl, 2019-11-20:
I've updated using your icon (though I exported it to the .icns format that MacOS needs). I renamed main.exe to WebControl-backgrounder for the main-osx.spec since it runs as a backgrounder in MacOS and that makes it show up next to the frontend 'WebControl' app in the ActivityMonitor process list.
 The master branch '810eb78' builds and runs fine. I'll IM you a link to the file on the Maslow forum.  I'll do the update when you post it, and send that link then.

blurfl, 2019-11-24:
While the MacOS app runs (mostly), the help menu update function is inoperative - the app shuts down. I'll have to dig into the updating mechanism to figure out what's going on.

madgrizzle, 2019-11-24:
Theres a few files that are not included in the tools directory that are needed and are specific to each operating system.  I'll get a list of them.

madgrizzle, 2019-11-24:
Well, realized as well that it might not work because I never developed code to test if its macos or not (not sure it's needed, but never know).

maybe I can get a vm up and running on macos, but in the meantime, when the program starts, it prints out a few lines (among many others) that indicates what the platform and install type.  It might be something like:

linux
-------------------
singledirectory
-------------------

Am I correct in that you can't see print messages (i.e., messages that only appear in the terminal window)?

blurfl, 2019-11-24:
Instead of 'linux', the platform line is 'darwin' and the install type is single directory

blurfl, 2019-11-24:
You're correct about the messages, I've tried in the terminal as well. Does the code require that it run from a directory named 'webcontrol' in the user's home directory? I haven't tried that, I'm working in my github desktop setup.

madgrizzle, 2019-11-24:
No, I think it can run anywhere.. but it has to be able to find the directory where it's run from to locate the script in the tools directory.  I had to create a special one for Windows.. not sure if I need to create a special one for macos.. I'll add a hook for Darwin and see if we can get it at least not crashing.

madgrizzle, 2019-11-25:
I'm not sure what needs to be changed to support updating webcontrol on macos.  If you look at the ReleaseManager, you'll see that it has two ways of doing the upgrade; one for windows and one for everything else.  Windows uses 7za to extract the downloaded file, but everything else uses tar.  Does macos support tar natively?  Look at the upgrade_webcontrol.sh file in the tools directory..

```
#!/bin/bash
sleep 3
echo $1
echo $2
cd $2
tar -zxvf $1
./webcontrol
```

does any of it choke macos?

madgrizzle, 2019-11-25:
And not related to updating webcontrol, to be able to update the firmware, you will need to include avrdude and avrdude.conf in the tools directory.

I've also included a copy of my build script for Linux (buildLinux1.sh) in the root.  This will show you how I set everything up to reduce the amount of typing.  I have similar scripts for the RPI, win64 and win32.  Make sure you have a directory named 'releases' in the root of your repo.. the script doesn't create it.