WebControlCNC/WebControl #49: WC hangs during startup

blurfl, 2019-11-21:
I noticed tonight that something recent has broken the master branch. After running pyinstaller on it, when it runs I get the following error in the terminal:
```
... Note: earlier startup lines removed ...
Sent Board Cut Data compressed
Exception in thread Thread-2:
Traceback (most recent call last):
  File "threading.py", line 926, in _bootstrap_inner
  File "site-packages/gevent/threading.py", line 177, in run
  File "threading.py", line 870, in run
  File "Background/messageProcessor.py", line 22, in start
  File "HelpManager/helpManager.py", line 21, in checkForUpdatedHelp
  File "HelpManager/helpManager.py", line 28, in getHelpPages
  File "site-packages/github/Repository.py", line 1514, in get_contents
  File "site-packages/github/Requester.py", line 276, in requestJsonAndCheck
  File "site-packages/github/Requester.py", line 287, in __check
github.GithubException.UnknownObjectException: 404 {'message': 'Not Found', 'documentation_url': 'https://developer.github.com/v3/repos/contents/#get-contents'}

```

 Using git bisect, I think the issue arose [in this commit].(https://github.com/madgrizzle/WebControl/commit/18964a199b7238cef94c5f1c566911b336f9eec5)

```
18964a199b7238cef94c5f1c566911b336f9eec5 is the first bad commit
commit 18964a199b7238cef94c5f1c566911b336f9eec5
Author: John Hogan <john@thehogans.name>
Date:   Wed Nov 20 12:50:17 2019 -0500

    work on help management

 Documentation/boardManagement.md     | 38 -----------------------
 HelpManager/helpManager.py           | 60 +++++++++---------------------------
 WebPageProcessor/webPageProcessor.py | 19 +++++++++++-
 docs/boardManagement.md              | 38 +++++++++++++++++++++++
 templates/helpPages.html             | 16 ++++++++++
 5 files changed, 87 insertions(+), 84 deletions(-)
 delete mode 100644 Documentation/boardManagement.md
 create mode 100644 docs/boardManagement.md
 create mode 100644 templates/helpPages.html

```

blurfl, 2019-11-21:
I think that [this]9https://github.com/madgrizzle/WebControl/commit/18964a199b7238cef94c5f1c566911b336f9eec5#diff-8bd85aed7ed115bdd56998131da38621) is it; the "Documentation" folder is renamed to "docs", helpManager.py line 28 needs to be updated.

madgrizzle, 2019-11-21:
Yeah, sorry about that.. not used to people actually working from my repo.  I've started working on something and then got called away and couldn't fix what I had done.  It should be 'working' now .. or at least not broken.

madgrizzle, 2019-11-21:
So what I'm trying to do is use github pages to host the help contents and then allow webcontrol to pull from there was well, as needed.  The thought is if all the help pages are stored in the 'docs' folder, then I can include them in a release and they would be available off-line in webcontrol (i.e. webcontrol not connected to the internet) and online if someone just browses to https://madgrizzle.github.io/WebControl.  What I was working on was, in webcontrol, was pulling the contents of the docs folder with the idea that if there is something newer in the repo, it would be pulled in and used by webcontrol instead of what was included in the release.  This would allow the documentation to be updated in the repo, but not require rebuilding a release just because documentation changed.  However, I'm not sure how I can tell if something is new or not, or even how to handle things like images.. but it was a start.

blurfl, 2019-11-21:
Keeping docs up to date would be valuable, but how to get the version of docs that pertains to the version the user is running? A good puzzle, keep on digging!

blurfl, 2019-11-21:
Your fix did the trick. 👍💯

closed 2019-11-21 by blurfl

davidelang, 2019-11-21:
On Thu, 21 Nov 2019, Scott Smith wrote:

> Keeping docs up to date would be valuable, but how to get the version of docs 
> that pertains to the version the user is running? A good puzzle, keep on 
> digging!

That would be my concern.

I've been bit too many times with docs that refer to features not in the release 
I'm running :-)

David Lang

madgrizzle, 2019-11-21:
I can add metadata to yaml portion of the file that includes the release version number, have webcontrol parse the file for that release version number, and if newer than the current release version number, then stick with the one that's in the release.  I might add an additional identifier to indicate the document's version number.. so something like:

>wcversion: 0.915.001

where 0.915 refers to the release version and .001 refers to the document version.  I was hoping to use the information that comes from the git rest api to determine if a file is newer than the one in the release, but I don't think its readily possible.  And I'm also not sure I could include new/edited images in the update unless I pull the replacement file directly from github pages and not from the repo... things to ponder.

But, for documentation, it might be a good idea to indicate in the documentation the minimum release required to support a particular feature.

madgrizzle, 2019-11-21:
.. something like

https://madgrizzle.github.io/WebControl/Boards/

madgrizzle, 2019-11-26:
I'm stuck on turning this into a release until I can figure out the pyinstaller problem with markdown extensions...

https://stackoverflow.com/questions/59052079/pyinstaller-and-python-markdown-importerror-no-module-named-extra

madgrizzle, 2019-11-26:
Well, looks like it might be an issue with a particular extension as I was able to get around the problem.. only took 4 hours to isolate it.

BarbourSmith, 2019-11-26:
It's a bummer to spend four hours on something which turns out not to be your problem 😔

madgrizzle, 2019-11-26:
I try to keep in mind that I probably would have spent 400 hours trying to write a routine that did what the library did, with only a fraction of the capability.. it's the price you pay.