WebControlCNC/WebControl #50: Make the single directory releases have a top level dir

davidelang, 2019-12-14:
It's bad for, to have a tarball release that puts lots of files in the dir you start in, everything should move down a layer

madgrizzle, 2019-12-14:
Not sure what you mean.

madgrizzle, 2019-12-14:
Oh, I think I know what you mean.. I think I had to do it that way for a reason, but I'll see if I can figure out a way not to have to.

davidelang, 2019-12-16:
from your other comments, I think it has to do with the difficulty is finding 
your parent directory (you may have several with symlinks)

so what I would suggest is try to make it so you have a webcontrol.libs 
directory and a webcontrol binary in the place where you are running. That 
should allow you to update everything without having to worry about parent 
directories and still avoid too much stuff if it gets untarred in a home 
directory.

David Lang

  On Sat, 14 Dec 2019, madgrizzle wrote:

> Date: Sat, 14 Dec 2019 13:01:16 -0800
> From: madgrizzle <notifications@github.com>
> Reply-To: madgrizzle/WebControl
>     <reply@reply.github.com>
> To: madgrizzle/WebControl <WebControl@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Author <author@noreply.github.com>
> Subject: Re: [madgrizzle/WebControl] Make the single directory releases have a
>      top level dir (#50)
> 
> Oh, I think I know what you mean.. I think I had to do it that way for a reason, but I'll see if I can figure out a way not to have to.
>
>

madgrizzle, 2019-12-16:
I've figured out how to find the parent directory, just trying to figure out how to accommodate current users without causing it to break on them.  I might have to put out an interim release to prepare for that  change.

> so what I would suggest is try to make it so you have a webcontrol.libs
directory and a webcontrol binary in the place where you are running. That
should allow you to update everything without having to worry about parent
directories and still avoid too much stuff if it gets untarred in a home
directory.

I have no idea how to do that with pyinstaller.

blurfl, 2019-12-16:
I found [this](http://unafaltadecomprension.blogspot.com/2014/07/pyinstaller-separating-executable-from.html) about putting libs in a sub directory, and a bit about the temp files left behind in the Windows  singlefile apps.

madgrizzle, 2019-12-16:
Thanks @blurfl, that helped a lot.  I think doing it that way will make it not break the current upgrade path.

madgrizzle, 2020-01-08:
That method doesn't work unfortunately.  Maybe did for earlier version with python2 perhaps.  Pyinstaller github still shows this as an active enhancement request.