WebControlCNC/WebControl #51: Goto line numbering off by -1 vs view/edit gcode numbering

gb0101010101, 2019-12-22:
1. Load up gcode
2. View/Edit Gcode using menu and find a line you want to go to.
3. Close out View/Edit
4. Use the Goto input box, enter the line number, click Go

The GCode displayed under the input box is off by -1 line in comparison to the View/Edit window.
e.g. View/Edit line number 500 is actually Goto input box line number 499.

madgrizzle, 2019-12-22:
Ah, I bet because of the list index.  Relatively easy to fix.  Thanks for finding.

thejobbitt, 2020-01-02:
I have had this issue as well but it is off by -2.

 I am also confused by the "Goto line number" function. I have tried running gcode from a line number where the machine left off, but it only runs the line number that is entered and maybe one or two more lines then stops. I'm assuming it is supposed to run through the rest of the file?

madgrizzle, 2020-01-08:
It's supposed to run through it all.. can you send the gcode to masgrizzlemaslow at gmail dot com?

gb0101010101, 2020-01-08:
> but it only runs the line number that is entered and maybe one or two more lines then stops. I'm assuming it is supposed to run through the rest of the file?

Had the same problem and figured it out - don't know if it can be easily fixed. Basically you cannot just open a gcode file, goto a line number far down in the file, and start from there. That is because the first few lines of gcode setup the machine and need to be run. What I did was add tool change directly after the first setup lines before any movement takes place. I then open the file, Run it, it pauses on the tool change, goto a line line number, then resume. 

Perhaps it would be possible for WebControl to detect the first few line of code, before a movement is made and ask if you want to run it before starting/resuming a file from a set number that is not the beginning of the file?

davidelang, 2020-01-08:
@gb0101010101

note that those 'first few lines' can appear again elsewhere in the file, 
changing the settings.

If the file does absolute mode, then it's fairly easy to jump around the file, 
but if it ever does relative mode, it gets really ugly to jump around.

you really need to have  g-code simulator to walk though the steps and track the 
state of everything so that when you want to jump around, it can set the 
appropriate state at that point and move to the proper place.

David Lang

madgrizzle, 2020-01-08:
Send me the file.. what you are describing is exactly what I tried to accomplish.

gb0101010101, 2020-01-08:
I have created a new issue #56 for "Cannot 'Play' gcode file from specific line number when using 'Goto' input box" as mentioned by @thejobbitt and also experienced by myself. Hopefully we can keep these issues separate. 

@madgrizzle I have sent email with gcode file that experienced both issues mentioned here.

madgrizzle, 2020-01-08:
Regarding index.. I've fixed the indexing so it corresponds with what you see in view/edit gcode screen.  A "problem" still exists in that webcontrol, which uses routines from ground control, modifies the file to clean it up.  It strips all blank lines, puts spaces between gcode commands, makes gcodes uppercase, etc.  So there won't be a 1:1 relationship with the original file.  Is it possible that the -2 issue could have been due to a blank line in the file being filtered out, @thejobbitt?

gb0101010101, 2020-01-12:
> A "problem" still exists in that webcontrol, which uses routines from ground control, modifies the file to clean it up.... So there won't be a 1:1 relationship with the original file.

The _View/Edit_ window is displaying the cleaned version, with blank lines stripped out, on version 0.918. If this is the same gcode that is used by the _Goto_ feature then I don't think there is a problem. It does not need to correspond to the original file just as long as the cleaned version is used everywhere in WebControl. It makes sense that the user should see the cleaned version in _View/Edit_ if that is the gcode that WebControl is using to drive the Maslow.

Thanks for the fix. I will test it out soon. Almost got my IDE setup.

gb0101010101, 2020-01-12:
Just pulled and tested v0.919. _View/Edit_ numbering now matches _Goto_ feature. Tested using a few long ~35K line files and it worked throughout as expected. 

This issue is resolved for me. Leaving it open so that @thejobbitt can confirm the fix.