WebControlCNC/WebControl #52: Play, Pause, Resume optimization

gb0101010101, 2019-12-22:
Fantastic work on this. Most functionality works great and is a great improvement over GroundControl. Submitting multiple issues whilst they are fresh in memory which have no reflection on your effort. Don't know how to categorize them as Feature Request vs Bug Report on GitHub.

**Primary Bug/Issue:**
_Play_ button can be clicked after it has already been clicked and has weird outcome. Using gcode from Fusion360 (with Maslow post processor) which always insert a tool change at the beginning.

**Reproduce:**
1. Load gcode
2. Click _Play_ button
3. Tool change request is displayed, click _Resume_
4. Click _Pause_ button before router head reaches the first XY cutting coordinates.

**Problem:**
1. Can click the _Play_ button again.

**Result:**
1. Tool change operation is run again non interactively. 
2. The gcode preview window shows this happening.
3. The Z-Axis plunges to stock bottom at current XY location then raises back up previous Z Depth. This cuts a full depth hole where it may not be intended.
4. No further gcode is processed and nothing happens after this. Clicking _Pause/Resume_ has no effect. Clicking _Play_ again repeats step 3  and 4.
5. The _Stop_ button must be clicked to get out of the cycle.

**Suggested quick fix:**
When _Play_ button is clicked then it should be disabled visually and functionally.

**Ideal fix**
The current _Play_, _Pause/Resume_, _Stop_ buttons are not ideal UX. 

1. _Stop_ should only be functional after _Play_ has been clicked.
2. _Play_ should be non functional after it has been clicked.
3. _Pause_ should only be functional after _Play_ has been clicked.

I suggest combining the _Play_ and _Pause/Resume_ buttons. There would only be two buttons to control operation:

_Play_ -> _Pause_- > _Resume_ [Loop: _Pause_ -> _Resume_]
_Stop_ - Always enabled.

It would cycle through the following states:
- _Play_ and _Stop_ buttons visible. _Play_ enabled, _Stop_ disabled.
- Click on _Play_ - Change _Play_ button to _Pause_ button. Enable _Stop_ button.
- Click on _Pause_ button (previously _Play_) - Change _Pause_ button to _Resume_ which is enabled. 
- Click on _Resume_ button - Change _Resume_ button to _Pause_ (which is enabled)
- Click on _Stop_ button (at any time) - Change _Play/Pause/Resume_ button to _Play_ and enable. Disable _Stop_ button. 

Which ever was you go I also suggest keeping the the buttons 'sticky' (always visible) at top of browser once the _Play_ button has been clicked. This is only a problem on small screens (phone/tablet) where these buttons can go off screen. I think these button are of primary importance when the machine is running (Play state).

Please let me know if I have not communicated this clearly; its been a long day and I wanted to log it whilst it was fresh in memory. Going to setup dev env and try to find/fix code but not great at cross compiling platforms and expect compiling on RaspberyPi to be very slow. Let me know how I can help and thanks for all the effort.

madgrizzle, 2019-12-22:
ANY help would be greatly appreciated.  I cannot really express how much it would be appreciated.  And I have no pride in authorship, so you don't have to be PC with comments.  Practically all of this has been a one man show in development by someone learning how to code python, fortunately I've had a lot of help with testing and stackoverflow.

 I'll dig through your comment later today, but I'll say that they layout was intended to mimic ground control to make it easy for people to transition to it.  Also, the stop button has proved handy in a few situations even when the machine is stopped and really does need to stay or a few more changes would be needed.  For example, a user presses the directional error to move the sled 2 feet left but wants to stop it mid-move.  The play button would also need to have been cycled to stop during this move so the user could stop it.  But then, my concern would be that the user is going to click the stop button just as the move completes and it changes to play right when he mouses down on the button.. now the machine starts running gcode..but the user wanted to do an emergency stop..  a solution could be to put a delay in responding to button press (one second maybe) to avoid that potential.  I try to follow Murphy's law (and all variants) when I program.

gb0101010101, 2020-01-12:
Yep see your point. The _Stop_ button should be enabled all the time.

gb0101010101, 2020-01-24:
Made a simple proof of concept test branch to address this issue:
https://github.com/madgrizzle/WebControl/compare/master...gb0101010101:gb0101010101-test-ui-states

Originally I was only thinking about _UI Buttons_ but now realize that most of _UI_ should be disabled once Maslow is running (state=Play). So _Menu_ should be disabled too. I have only done _UI Buttons_ in this test branch.

Test code is only using Client side JavaScript to Enable/Disable buttons. It should have Server side code to validate client inputs.  

There are 4 states tracked:
- Play
- Stop
- Pause
- Resume

In HTML templates buttons are assigned CSS classes _state-play_ and _state-pause_. 

_state-play_ CSS buttons are disabled when _Play_ button is clicked.
_state-pause_ CSS buttons are enabled when _Pause_ button is clicked.

_Pause_ button is disabled by default.
_Stop_ button is always available.

When:

- _Play_ button is clicked all _state-play_ buttons are disabled.
- _Pause_ button is clicked all _state-pause_ buttons are disabled.
-  _Resume_ button is clicked all _state-pause_ buttons are enabled.
-  _Stop_ button is clicked all  _state-play_ and _state-pause_ buttons are enabled.

gb0101010101, 2020-01-24:
Note: Issue #85 interferes with this and stops it from being fully functional.

madgrizzle, 2020-01-24:
This was something I had fiddled with and I like what you've done.  Always keep in mind that webcontrol has to tell the clients, on a regular basis, what the current state is (I know this is just a concept at this point).  If someone is running a browser on their computer, pauses it, and then loads up a browser on their phone, webcontrol needs to let that browser know to change the uistate.

gb0101010101, 2020-01-25:
@madgrizzle; I think I have this working. Its doing 2 way communication and updates button states on all attached clients.

Check it out and let me know if you have any problems or if you want me to change anything. If its good I will create a pull request.

https://github.com/madgrizzle/WebControl/compare/master...gb0101010101:gb0101010101-test-ui-states