WebControlCNC/WebControl #53: Z-axis modal UI window: additional "Zero", "Traverse" buttons. "Touch" button? [feature request]

gb0101010101, 2019-12-22:
When the _Z-axis_ button is pressed, and the z-axis modal window is displayed, it would be great if there were two more buttons:

"Go to Zero"  - Move z-axis to current zero (0) height. Helpful when aligning a different router bit when a tool change is requested. 
"Go to Traverse height" - Move z-axis to safe traverse height (above stock). Helpful when skipping though gcode to resume cutting after a failure. 

There is currently a _Touch_ button which is always seems to be disabled and therefore does nothing. Is this supposed to move z-axis to 0?

Again: Fantastic work, let me know how I can help. Can make icons for these operations if you want to minimize button area on screen.

madgrizzle, 2020-01-27:
I don't have a system in place to really test the touch plate.  Is this something you can work on.  I think it would be good to get this sorted out.

gb0101010101, 2020-01-27:
Don't have access to the touch senor but will do the other buttons once the positioningMode has been fixed.

gb0101010101, 2020-03-02:
Created pull request https://github.com/madgrizzle/WebControl/pull/108

thormj, 2020-07-13:
If it's merged... do I need to do anything to see the new buttons?