WebControlCNC/WebControl #54: Fool WebControl into thinking Maslow control board is present [Question]

gb0101010101, 2019-12-22:
I predominately write code at location A when MaslowCNC is at location B. This causes a problem because the location A RaspberryPi  is not connected to the location B Maslow board. Is there a way to fool WebControl into thinking that the Maslow board is present and responding for development?

madgrizzle, 2019-12-22:
The Arduino accepts a special command to put it into "fake servo" mode where you don't need motors connected to it and it behaves as if it does (I.e., no errors).. but there's nothing in webcontrol software that would allow you to develop/troubleshoot controller interactions without an Arduino connected.  I bought a couple cheap mega arduinos that are plugged in to a couple development machines for that purpose.

gb0101010101, 2020-01-12:
Great thanks.

closed 2020-01-12 by gb0101010101