WebControlCNC/WebControl #55: Test Rig

thejobbitt, 2020-01-02:
Could you share with me how you have your test rig setup? I'm thinking about building one to help with development.

madgrizzle, 2020-01-03:
I just have an rpi plugged into an arduino with it set to fake servo mode.  Nothing special.

Most of my development is done on my win10 laptop running pycharm as my idea.  I also have a spare Arduino connected to it.

gb0101010101, 2020-01-12:
@madgrizzle, Do you have the Maslow board plugged in to Arduino or can it run without it?

madgrizzle, 2020-01-12:
I have a board plugged in set for fake servo.. it'll startup and such without it, but you can't do much without an Arduino connected.

blurfl, 2020-01-12:
> Do you have the Maslow board plugged in to Arduino

You don’t need a Maslow board to use the ‘Fake Servo’ mode. The firmware will respond as if a v1.2 Maslow board were present.

gb0101010101, 2020-01-12:
OK Great. Will dig out a Mega 2560 and try it.

For others: To enable Fake Servo
- Menu: _Actions_ -> [scroll down to] _Fake Servo (Advanced)_
- Enter the secret code and click _Enable Fake Servo_

The secret code appears to be the answer to _Ultimate Question of Life, the Universe, and Everything_ (Haha - Love it). Can anyone confirm? Could not find it documented anywhere.

madgrizzle, 2020-01-12:
You've got it.. I'd rather people ask for the secret or be smart enough to know to figure it out than to make it easy and someone unwittingly enter fake servo mode.