WebControlCNC/WebControl #56: Cannot 'play' gcode file from specific line number when using 'Goto' input box.

gb0101010101, 2020-01-08:
I had an issue with cutting and decided to call it quits for the day so clicked the 'Stop' button. Next day I wanted to restart cutting from a specified section/line number of the same gcode file.

I tried:
1. Load gcode file
2. View/edit gcode file to find the section/line number I wanted to restart from
3. Enter the line number into the 'GoTo' input box.
4. Click 'Play' button top start cutting.

This did not work with the following happening: 
1. The Left/Right motors made a continuous noise as if they were moving but there was no apparent movement or it was so small that I could not notice it. 
2. Clicking on 'Pause' button had no effect with continued motor noise and not movement but did display 'Resume' button.
3. Clicking on 'Resume' button had no effect on motor noise and still no movement. 
4. Clicking on 'Stop' did stop motor noise.

To get it to work I did the following:
My gcode is from Fusion 360 using Maslow Post Processor. This by default includes a tool change after initial setup lines.
1. Load file.
2. Click 'Play' button.
3. Tool change requested automatically.
4. Find section/line number and enter it in GoTo input box.
5. Click 'Resume' button.

davidelang, 2020-01-08:
If the maslow was powered down in the meantime, when you tried to restart you 
have a bunch of initialization that did not happen, possibly including setting 
the feed rate.

we should look at modifying the firmware to remember mode and feedrate info like 
it remembers the position. That would help a lot of things.

Either that or we need it to generate an error message if these things are not 
set before you hit a move command.

David Lang

gb0101010101, 2020-01-08:
Damn @davidelang you are fast. Trying to write comments and emails on other issues to direct here and you have already responded! Just figured out I created this issue wrong and should have used 'Reference in new issue' feature.

So I @gb0101010101 said in #51
> Perhaps it would be possible for WebControl to detect the first few line of code, before a movement is made and ask if you want to run it before starting/resuming a file from a set number that is not the beginning of the file?

And you said:
> note that those 'first few lines' can appear again elsewhere in the file,
changing the settings.
> If the file does absolute mode, then it's fairly easy to jump around the file,
but if it ever does relative mode, it gets really ugly to jump around.
> you really need to have  g-code simulator to walk though the steps and track the
state of everything so that when you want to jump around, it can set the
appropriate state at that point and move to the proper place.

So yes - I agree with everything. This is why I thought it:

1. Would not be an easy fix. 
2. Would require confirmation from user before detected gcode is executed. 

A gcode simulator would be an ideal solution.

gb0101010101, 2020-01-08:
>  we should look at modifying the firmware to remember mode and feedrate info like it remembers the position. That would help a lot of things.

Don't think that would be feasible unless we could also track the project/job/gcode file. Don't want to retain any settings when the project changes and it might be even more difficult to correctly track that. It would be great if WebControl could do this.

davidelang, 2020-01-08:
On Tue, 7 Jan 2020, gb0101010101 wrote:

> Don't think that would be feasible unless we could also track the project/job/gcode file. Don't want to retain any settings when the project changes and it might be even more difficult to correctly [...]

The firmware doesn't know anything about project/job/file so we can't do that.

But it's normal for CNC machines to retain their state, and a 'good' gcode file 
should set the units, absolute/relative, feed rate (g0 and g1), so if you are 
feeding it 'good' files it won't matter what is remembered.

But when you send it either raw g-code, or are picking up a partial cut, it 
would be better to remmber the state and keep using it rather than be in an 
undefined condition of no state being defined.

the other option would be to define what state these settings should be in at 
startup, but that's also going to be wrong a fair percentage of the time, so I 
think a 'remember the last state' is more useful.

David Lang

madgrizzle, 2020-01-08:
I believe the issue might be related to the tool change operation at the beginning of the file, and not the processing of the gcode.... I initially loaded the file, went to line 1003 and hit play and it worked for me.  But when I stopped it, and hit play, it didn't actually do 'play' it (entirely).  It did send the T1 M6 command to the controller, but since the controller already had T1 installed (because the controller keeps track of current tools and I had already sent that command when initially running the file), it never sent a 'Tool Change:" message to webcontrol in order to prompt the user to change the tool and changes the pause button to resume.  This only happens when it sees a 'Tool Change:' message.  This wouldn't necessarily be a problem, but a couple months ago, we discovered tool change was not working because of something in the stock firmware and, long story short, I made a change in webcontrol to pause sending gcode when webcontrol detects its sending a tool change.  The only way to unpause webcontrol when this happens is to wait to get the 'Tool Change:' message from the controller.  Since this is not sent when the tool of the tool change command matches what the controller is tracking as the current tool, that message is never sent.. therefore webcontrol sat there not sending gcode and the pause button on the UI was never changed to 'Resume'.  Basically, there was no way to get webcontrol out of that mode without hitting play... and hitting play actually causes issues if you are sending gcode or paused sending gcode.

I'm not sure the correct way to handle this but what I did... When webcontrol detects that it's sending a tool change message, it goes ahead and sends it AND changes the pause button to resume.  It isn't ideal because all that happens is the yellow pause button changes to a blue resume button, but there's no popup or anything... but if you notice your machine not doing anything, maybe you'll notice the resume button and text on the controller messages saying 'tool change to tool x' (the message the controller sends back saying it has changed the tool tracking number).  I *think* the only way to really get around this is by adding another message of some sort from the firmware OR tracking the current tool within webcontrol as well.  This latter *might* be the best way to handle it, but I'm not sure what problems that could introduce.

madgrizzle, 2020-01-08:
Ok, wasn't too terribly hard to track tool change in webcontrol.  There's no current way to know wat tool the controller has in its memory so we just have to hope webcontrol and the controller stays in sync.  I'll push a new release sometime today/tonight with this and a few other bug fixes.