WebControlCNC/WebControl #57: Long gcode lines cause right side bar to resize on PC

gb0101010101, 2020-01-12:
Accessing WebControl UI on Linux PC @ 1080P using Firefox 72.0.1. On my screen this makes the _controllerMessage div_ about 410px wide which allows for ~ 33 characters in line length. When the gcode text line exceeds this it causes the _div_ to resize continuously to fit the text. This in turn causes the whole _Right Sidebar_ to resize continuously making it sometimes difficult to click buttons as they change position.

The simple fix is to add CSS that will make the gcode break at any point (including splitting words/numbers) and wrap it to a new line:
`#controllerMessage {
  overflow-wrap: anywhere;
}`

This may not be an ideal solution as it may cause confusion reading the gcode but at least it stops the resizing problem and keeps all the gcode visible. It may be better to provide X overflow scrollbar which will retain proper formatting but only display part of the gcode.

madgrizzle, 2020-01-12:
Cool! I'll give it a try.  Im terrible at css.. thanks!

gb0101010101, 2020-01-23:
This appears to have been fixed/applied in 8c8455360502a24e2c3c731400caa960535e0f7c so closing. Thanks.

closed 2020-01-23 by gb0101010101