WebControlCNC/WebControl #60: Tracking current G90 (Absolute) and G91 (Incremental) mode?

gb0101010101, 2020-01-14:
Does WebControl track the current positioning mode G90 (Absolute) vs G91 (Incremental)?

Looking through the code I cannot seem to find this. If this is not tracked then we may have a problem. 

**Gcode:**
```
(TEST_POSITION)  
G0 G40 G90  
G21  
G00 X0 Y0  
(RELATIVE)  
G91  
G00 X150 Y150  
(PAUSE_HERE_AND_MOVE_Z)  
G00 X20 Y20
```

**Problem:**
1. Run gcode above all the way through. The final position should end up at X 170 Y 170.
2. Now run gcode again. 
3. When you see the G00 X150 Y150  command sent click _Pause_ button before movement completes.
4. Click _Z-Axis_ button to bring up the modal. Move z-axis up/down any amount.
5. Click _Close_ button.
6. Click _Resume_ button.
7. Final position is X 20 Y 20 which is wrong.

This is because the z-axis move action sends gcode 
`"G91 G00 Z" + str(float(distToMoveZ)) + " G90 "`
The G90 at the end incorrectly sets positioning mode to Absolute when the gcode file is in G91 (Incremental).

**Fix:**
The positioning mode (G90/G91) has to be tracked by WebControl and any action that changes it temporarily must revert it back to what it was.

madgrizzle, 2020-01-14:
I think you're correct.  It is supposed to track it when using index moves but I don't think I ever added anything for pause (certainly not in ground control).  I'll look into it, shouldn't be incredibly hard  (knock on wood)

gb0101010101, 2020-01-14:
My thought was that any time a line of gcode is sent to controller then it should be scanned to find last index of 'G90' or 'G91' and set that value somewhere that is globally accessible (self.data.positionMode?)

e.g. 
`G91 G00 Z5 G90`
Would save "G90" to self.data.positionMode

`G91 G00 Z10`
Would save "G91" to self.data.positionMode

Then in any _Action_ you could
```
positionModeOrg = self.data.positionMode

# Run gcode that changes position mode
 self.data.gcode_queue.put("G91 G00 Z" + str(float(distToMoveZ)) + " G90 ")

# Revert position mode
if positionModeOrg != self.data.positionMode:
  self.data.gcode_queue.put(positionModeOrg)
```

madgrizzle, 2020-01-21:
I've done a few things with this.  I think it now tracks during pause and will return to the correct state.  Please check if you would.  Also, I added a UI line to the frontpage to show the current tool as well as positional state (Feel free to make it look better!).  The update to those are sent whenever either value changes and, at a minimum, every 5 seconds when health messages are sent.. this allows new clients to get synced.  I also chopped up the frontpage button move commands into individual lines (i.e., "G90", "G0 X..", "G91").  In doing so, the UI will be up to date while the move is taking place.  If someone sends code as a single line (like the move used to do), it will just track the last positional value found in the line.

gb0101010101, 2020-01-22:
Thanks for spending time on this. Had unexpected results in testing with restored positioning mode so created a pull request with fixes: https://github.com/madgrizzle/WebControl/pull/81

madgrizzle, 2020-01-22:
See my comments to the PR.