WebControlCNC/WebControl #63: Motor Offset Height in MM Incorrect Update after Triangular Calibration

idougal, 2020-01-16:
I measured and entered the Motor Offset Height in MM to be 1069mm.

I ran the triangular calibration in Webcontrol and clicked update.  I immediately started receiving error messages that the sled couldn't be found.  Everything worked perfectly up to that point.

I started reviewing the settings to see what could have gone wrong and noticed that the offset height changed to 468.4.  I'm not sure where that new number came from, but it must have been modified by the triangular calibration update button.

madgrizzle, 2020-01-16:
The triangular calibration routine adjusts that parameter during it's optimization.  But 1069 mm seems really, really high.. you have a non-standard layout?

idougal, 2020-01-16:
It is very high
[image]

madgrizzle, 2020-01-16:
Wow.  Yes, very high.

davidelang, 2020-01-16:
I wonder if there is any code to throw out 'insane values' that may be getting 
triggered here.

David Lang

idougal, 2020-01-16:
My wall wasn't long enough for the 12' span and I don;t want to hit me head
going through the door. :)

davidelang, 2020-01-16:
check the spreadsheet, with that much height, you may want to add a bunch of 
weight to the sled

David Lang

idougal, 2020-01-16:
> check the spreadsheet, with that much height, you may want to add a bunch of weight to the sled David Lang

Thanks for the reminder.  I did use the spreadsheet before I started building the frame to make sure I didn't get any red, but that was a year ago.
[Maslow Chain Geometry diff.xlsx](https://github.com/madgrizzle/WebControl/files/4069239/Maslow.Chain.Geometry.diff.xlsx)
  I just went through it again.  here are the numbers.  Is there anything to worry about?

idougal, 2020-01-16:
> check the spreadsheet, with that much height, you may want to add a bunch of weight to the sled David Lang

I'm going to flip the board and try the holey calibration tomorrow.  Should I adjust anything before that?

My SVG to Easel to Gcode circles look great, but I tried a couple of easel to gcode circles and they are  definitely warped.  The text carving was perfect on the same cut.

The first one was probably because the depth of cut was too deep, but the second one had identical settings as my svg to easel to gcode.

davidelang, 2020-01-16:
On Wed, 15 Jan 2020, idougal wrote:

>> check the spreadsheet, with that much height, you may want to add a bunch of weight to the sled David Lang
>
> I'm going to flip the board and try the holey calibration tomorrow.  Should I adjust anything before that?

I would say that we would want you to try it without changing anything so that 
we have good apples-to-apples comparison between the two.

then you can test again after changes.

David Lang

madgrizzle, 2020-01-16:
New release to fix that issue is forthcoming.

idougal, 2020-01-16:
> <author@noreply.github.com>
> Update after Triangular Calibration (#63)

You were in a discussion with someone on the forum where one of you was 
theorizing that higher would be more accurate.

madgrizzle, 2020-01-21:
I'm going to close this issue since its related to another..

closed 2020-01-21 by madgrizzle

idougal, 2020-01-23:
Looks good in 922