WebControlCNC/WebControl #64: Triangular calibration changes values not reported by traiangular calibration

idougal, 2020-01-16:
I manually entered all of my measurements into webcontrol and ran the triangular calibration.

It reported the following values:

Height of Motors = 1063.2
Rotational Radius: 146.6
Chain Sag Correction: 20.13201

I went back and reviewed all of the settings after clicking Accept results and the following settings were changed:

Distance between motors: I entered 3719.6 (actual gearbox to gearbox minus 40.4). The value changed to 3601.2.  Ground control measured it to be 3709.57 and 3711.57.  

It is extremely unlikely to be around 3600mm.  Why did that value change?

The motor offset height changed from 1056.3 to 468.4, even though the reported triangular calibration measurement was 1063.2.  I entered this as a separate ticket (#63 https://github.com/madgrizzle/WebControl/issues/63) because there is definitely something wrong with this part of the update.

The rotational radius didn't change from the default 128.1 value, even though the triangular calibration reported the rotational radius to be 146.6.

The Chain Sag Correction Value for Triangular Kinematics was set at the default 33 before the calibration.  The triangular calibration reported an actual chain sag of 20.1302.   

It did not change after the update.  Shouldn't it have?

I'm not sure if this is the correct behavior or not, but it doesn't make sense from an outsiders perspective.  I wouldn't have noticed if the issue from ticket 63 hadn't happened.

madgrizzle, 2020-01-16:
So distance between motors and motor offset changed but rotational radius and chain sag did not?  You're using stock firmware, correct?

madgrizzle, 2020-01-16:
And motor offset changed to some number that didn't appear on the screen..

madgrizzle, 2020-01-16:
Did you record the cut measurements you made?  If so, please post them and the original values of those four parameters when the cut was made.

davidelang, 2020-01-16:
please try webcontrol and the holey calibration option.

The triangular calibration option has you make a few cuts and then it tries to 
change the values to make the cuts come out right

note that you have to enter the distance from one of the cuts to the top of the 
workpiece, not to where the motors are for both calibration methods (they also 
want to know the height of the motors from the workpiece, but that's a separate 
value)

Calibration has turned out to be a _really_ hard task. There are lots of 
possible sources of error, and the resulting error is not linear to the 
contruction/measurement error, the errors are curves, and different errors cause 
different curves (some countering each other. 
https://forums.maslowcnc.com/t/list-of-sources-of-error/7523

Calbration is attempting to figure out what measurement is wrong and correct it, 
but it only has a few samples to work from and doesn't even try to account for 
most sources of error.

holey calibration has quite a few more measurements to work from and takes into 
account more error sources, but we don't have enough experience with it to make 
it the default yet (but just about everyone who's tried it gets better results 
than with normal calibration)

David Lang

idougal, 2020-01-16:
> So distance between motors and motor offset changed but rotational radius and chain sag did not? You're using stock firmware, correct?

Yes.  I believe I haven't updated the firmware to Holey yet.  That was going to be my next step.

idougal, 2020-01-16:
> And motor offset changed to some number that didn't appear on the screen..

Yes.  I entered 1056.3, calibration reported 1057.6, final number was 468.4

idougal, 2020-01-16:
> Did you record the cut measurements you made? If so, please post them and the original values of those four parameters when the cut was made.

Top cuts 1919.5

Bottom cut 1924

From top 268.  My board on that side might have been low.  At some point , that side of the board dropped 6mm, or so.

Bit was 0.1875

madgrizzle, 2020-01-16:
I'll look at it tomorrow to try to see if I can duplicate it.

madgrizzle, 2020-01-16:
Easy enough to duplicate it.  Somewhere along the way the html file for triangular calibration got messed up and called the wrong function on accept (It called the holey calibration accept function, which saved holey calibration default values and not the new triangular calibration settings).  Thanks, and sorry for the inconvenience.  Sometimes I get interrupted when modifying things and don't realize I didn't change something I needed to.  I'm making a new release now.

idougal, 2020-01-16:
No Problem.  I'm just glad people like You and Dave are keeping this
project going!

So when the new release comes out, Change all my settings back to original,
go to triangular calibration, enter the measurements, and then press
calibrate?

madgrizzle, 2020-01-21:
Please let me know if the fix worked.  If you just moved on to holey calibration, that's fine as well.. just let us know you can't readily test it and I'll close the issue.. I'm pretty sure it works, just want to leave it open to make sure I didn't screw it up.

idougal, 2020-01-23:
Everything looks good in 922

closed 2020-01-23 by idougal