WebControlCNC/WebControl #65: Unable to Open G-Code from Open Gcode Menu

idougal, 2020-01-16:
webcontrol-0.919-win64

If I Click G-Code / Open: I see Select Directory and Select file.

Select Directory has ./ in the pull down list. If I click the arrows, no other options appear like it does in ground control.  

I can't open any directories and files.

davidelang, 2020-01-16:
you first have to upload the file to the Pi, then you can open it.

I know, it's a bit confusing, but that's the current process.

David Lang

idougal, 2020-01-16:
OK.  It sounds like I'll be filling up the memory if I don't delete the
files, which I don't know how to do yet. (This is all new to me)

I'll keep copying and pasting the gcode for now.

Thanks for the help.

davidelang, 2020-01-16:
On Wed, 15 Jan 2020, idougal wrote:

> OK.  It sounds like I'll be filling up the memory if I don't delete the
> files, which I don't know how to do yet. (This is all new to me)

there isn't a way to delete files through the web interface right now, but most 
g-code files are really small, so you would need to upload thousands of them to 
make a noticable dent in the available storage, your SD card is several GB in 
size and most files are a few KB in size. (although, you would get confused as 
to what's where long before that, so it is something we need to improve, but 
it's not a high priority)

David Lang

madgrizzle, 2020-01-19:
Closing this.

closed 2020-01-19 by madgrizzle