WebControlCNC/WebControl #71: Webcontrol Holey Calibration Accept Results button doesn't appear to do anything

idougal, 2020-01-18:
webcontrol-0.920-win64-singlefile.zip

- I pressed the "Upgrade Holey Firmware:51.27 button.
- I ran the Holey Configuration.
- I entered the measurements 
- I pressed Calculate and the calculations created the following numbers:
Motor Height = 1044.66923
Motor Distance 3674.95128
Left Chain Tol = -0.21402
Right Chain Tol = -0.18505
Cal Error: 2.353674533147113
- I pressed accept results
I went back to Settings/Maslow Settings and the following results are listed

Distance  = 3719.6
Motor Height = 1056.3

Then I went to Settings / Advanced settings and the following results are listed:

Chain Tol Left = 0
Chain Tol Right = 0

Shouldn't these number change when I press accept results?

madgrizzle, 2020-01-18:
Gives you a lot not confidence, huh?  I look at it in the morning.. something is up for certain.

madgrizzle, 2020-01-18:
Yep, didn't do anything.  Problem was that I moved the accept button to be on the same line as the calculate button.  That put the button within the form and therefore, for whatever reason, the javascript on-click code doesn't run (any idea why?).  I must have forgotten about that when I made the change (made the change so people readily see the accept button).  So I moved the button back to where it was (at the very bottom) and it works now.  I'll make a new release today/tonight with this change and the ui changes incorporated.

madgrizzle, 2020-01-21:
Let me know if this fixed it so it can be closed.  Thanks.

idougal, 2020-01-23:
It did work in 922, but the calculated numbers were slightly different

Pre 922 | 922.00000
-- | --
Between motors 3677.413 | 3768.75406
Height 1044.854 | 1044.33488
  |  
  |  
  |  
  |  
  |  
  |  
  |  
  |  
  |  
  |  
Left Chain -0.15281 | -0.17030
Right Chain -0.17296 | -0.16872

davidelang, 2020-01-23:
when youdo holey calibration multiple times, each time it works from the prior 
one

and it's sensitive that even a slight error in your measurement will change the 
results.

David Lang

idougal, 2020-01-23:
I made sure all the settings were identical to my pre test settings before I ran the actual calibration previously.  Then I typed in the same measured values from that test that I had typed in before.

Shouldn't identical settings and measurements output the same calculated results?

davidelang, 2020-01-23:
On Wed, 22 Jan 2020, idougal wrote:

> I made sure all the settings were identical to my pre test settings before I 
> ran the actual calibration previously.  Then I typed in the same measured 
> values from that test that I had typed in before.
>
> Shouldn't identical settings and measurements output the same calculated 
> results?

including identical calibration values for each run?

there is an ability to backup your json settings file and restore it, that's the 
only way to be very sure that you are at the same starting point.

David Lang

madgrizzle, 2020-01-23:
If you have the exact same settings as previous and enter the exact same measurements, you should get the exact same result.  I tested this by pressing calculate two times and the values didn't change the second time around.  But, even a slight change in setting could yield different optimized values.  I will say that I made no change in 922 that would have caused different results from 921.  It was just an simple html file adjustment.

davidelang, 2020-01-23:
On Thu, 23 Jan 2020, madgrizzle wrote:

> If you have the exact same settings as previous and enter the exact same 
> measurements, you should get the exact same result.  I tested this by pressing 
> calculate two times and the values didn't change the second time around.

right, but if you save your settings and then go back to the calibration and 
re-enter the same values, you aren't in the same starting point. (I think you 
understand this, but it's an easy trap for people to fall into)

David Lang

madgrizzle, 2020-01-23:
I assumed he didn't save his settings because the button to accept the setting changes wasn't working... but yes, you are correct in how it does work.

idougal, 2020-01-23:
The test settings under After test 3 were the settings in the machine before I ran my big calibration test.
[image]
I was pretty happy.  Straight lines look straight, squares look like squares, circles look like circles.  Most errors were less than 2mm and the biggest error was 5mm.

[image]

Then I started testing previous tickets.  Then I was testing this ticket and went back into the settings and made sure all settings matched "before Test 3" above.  
[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Then I pretended to run the holey calibration and entered the same measured values after Test 3 ran:
[image]

[image]
[image]
[image]

When I click calculate, I am expecting to see the same calculated values I got when I ran the test

[image]

But I get these values instead

[image]

I accepted them and just ran the top line of my test cut and I get a u-shape instead of the straight line I got earlier.

[image]

I tried manually putting all the settings in that I ran the test cut at and ran it again. The new line isn't quite as u-shaped, but it's not straight and nowhere near the original cut with what should be identical settings because I'm manually changed them all and double checked them.

madgrizzle, 2020-01-23:
Thanks for all the detail.  I put your numbers into my copy of webcontrol running 0.922 (matched your settings and measurements) and hit calculate and got the exact same numbers you did...

[image]

Then I did 0.921 and got the same results

[image]

Then I did 0.920 and got the same results

[image]

I'm fairly confident nothing changed in the computations because I haven't messed with that in a long time.

gb0101010101, 2020-02-10:
> @madgrizzle That put the button within the form and therefore, for whatever reason, the javascript on-click code doesn't run (any idea why?)

The button was probably submitting the form. You can typically stop this by adding `; return false;` to the onClick.

madgrizzle, 2020-02-18:
Closing as I think it works now.

closed 2020-02-18 by madgrizzle