WebControlCNC/WebControl #72: Add ability to export configuration to groundcontrol.ini

amantalion, 2020-01-19:
Hi @madgrizzle, thank you for the amazing work! Can I ask you to add one more feature, please? Can you add an ability to export configuration back to groundcontrol.ini (so that webcontrol and groundcontrol can be used interchangeably)?

madgrizzle, 2020-01-21:
**Note, nothing here is meant harshly.. **

I'm hesitant about the effort to take it on.  WebControl has most, but not all, of the settings that GroundControl has and therefore I can't readily generate a new groundcontrol.ini file.. I'd have to make the routine request the user to upload a groundcontrol.ini file, modify the settings in that file to equal that of webcontrol's, and then download the file back to the user.  That's a lot of work when the only values the values really shouldn't change often.. maybe just calibration?  And honestly, its not really hard to just write them down from webcontrol and enter them into groundcontrol.

amantalion, 2020-01-21:
@madgrizzle Ok, thank you. That was just a suggestion. Yes, I think it is easy enough to modify the groundcontrol.ini file with values from WebControl.

closed 2020-01-21 by amantalion

davidelang, 2020-01-21:
I would say that when inporting, anything that webcontrol doesn't understand 
should be saved as a comment (possibly save the entire file) and when writing it 
out, write the original file with values replaced

if there is no original file, either report that you can't export it reliably, 
or export default values (maintinance headache as default values will now exist 
in two places)

as stated, this is a nice-to-have, and if we end up migrating to WC as the 
default, the need will disappear in time.

David Lang