WebControlCNC/WebControl #73: webcontrol-0.921-win64-singlefile Doesn't Open

idougal, 2020-01-19:
I downloaded and unzipped webcontrol-0.921-win64-singlefile.zip

I double click on webcontrol.exe

The command prompt window opens, some text flashes by and the window closes.  I tried running as adminstrator.  Same result.

webcontrol-0.920-win64-singlefile.zip Opens fine with no problems.

I tried webcontrol-0.921-win64-singledirectory.zip.

Nothing happened initially, so I logged out and back in again.

Same result.  The command prompt window opens and is blank for a while.  Then some text flashes by and it closes.

madgrizzle, 2020-01-19:
yep.  I noticed something on the RPI build as well before I posted it and I had to rebuild it.  Didn't change anything, just had to rebuild.  Looks like maybe something just didn't work on this one.  I'll rebuild them all to be safe.

madgrizzle, 2020-01-19:
oops.. didn't mean to close it.

madgrizzle, 2020-01-19:
I noticed what might have been the problem.. I might not have loaded the virtualenv when I logged into the win64 build machine and so that would explain why it didn't build correctly.  I'll post a new file.

madgrizzle, 2020-01-19:
I rebuilt the Win64 files.  please let me know if it solves the problem (runs on my computer)

idougal, 2020-01-20:
Works now.

Thanks.

closed 2020-01-20 by idougal