WebControlCNC/WebControl #78: WebControl not opening Gcode files properly

amantalion, 2020-01-21:
Hi, has anyone had problems with their Gcode files when opening them with WebControl? It looks like my Gcode is missing some sections. Please, see the screenshot below to understand what I mean (it is supposed to be a circle within a square:

[image]

GroundControl is opening the Gcode file properly:

[image]

madgrizzle, 2020-01-21:
Can you email the file to masgrizzlemaslow at gmail dot com?

amantalion, 2020-01-21:
@madgrizzle Thank you for the fast reply. Just emailed you the file.

madgrizzle, 2020-01-21:
Well, that's the first time I've seen gcode lines without a gcommand.  I guess if it's the same command as previous line, it can be left out and still be valid gcode.  I'll have to work on that, but in the meantime, if you edit the code and fill in where the G command is missing, it'll display properly.

madgrizzle, 2020-01-21:
Are you able to view the gcode after its loaded in Ground Control and tell me if it looks different.. like if ground control added G0, G1, G2, etc. in front of the lines that didn't have them?

madgrizzle, 2020-01-21:
Nevermind.. that helped me solve a mystery I had long forgotten about.  Simple fix.  It will be updated in the next release.  Thanks!!

amantalion, 2020-01-21:
Great, glad that it helped! 

Yes, it looks the same in GroundControl, if I insert G commands from previous lines. Also, the problem was fixed in WebControl after I added those missing G commands.

amantalion, 2020-01-21:
@madgrizzle Also, do you know when the next release will be out approximately?

madgrizzle, 2020-01-21:
Probably tomorrow.

closed 2020-01-21 by amantalion