WebControlCNC/WebControl #79: defineHome in actions.py needs default values of "" specified for posX and posY

Orob-Maslow, 2020-01-21:
I found one issue with the actions.py file

    self.data.actions.defineHome()
TypeError: defineHome() missing 2 required positional arguments: 'posX' and 'posY'

(Line # 296 should be changed from)
>def defineHome(self, posX, posY):

 to

>def defineHome(self, posX = "", posY = ""):

Without the default value of nothing, calling defineHome with no posX or posY values fails because a value is required.  According to the comments, the function should use the current sled position if one isn't specified, but only if there is a default value, now "" with the modified line above.

madgrizzle, 2020-01-21:
Thanks!  I'll look at it in the morning.

madgrizzle, 2020-01-21:
It wasn't really a problem because the current functions that call defineHome always sends posX, posY (they're either "" or some value), but it certainly doesn't hurt to add the default values and is probably better form.

closed 2020-01-21 by madgrizzle