WebControlCNC/WebControl #80: The sled hits the bottom support beams during the Holey calibration

amantalion, 2020-01-21:
Hi, during the Holey calibration my sled hits the bottom support beams (on the photo below). I measured and entered correct values in Quick Configure. Has anyone had this issue before? If so, how did you fix it? Should I just manually change the vertical motor offset to make it work?

[image]

madgrizzle, 2020-01-22:
What values did you enter for the quickconfigure.  If you are too far off for any of them, it's possible for the controller to calculate too poorly..

amantalion, 2020-01-22:
Below are the values that I used.

**Kinematics Kit**: Ring
**Chain Around Sprocket:** Bottom
**Distance Between Motors:** 2994.12
**Height of Motors Above Workspace:** 430.0

BarbourSmith, 2020-01-22:
Those cuts don't look that close to the edge, is it possible that everything is working normally on the software side and this is something we should address on the hardware side?

I usually do my test cuts on a piece of scrap wood which is flush with those pegs the wood rests on so that the sled won't hit them

amantalion, 2020-01-26:
Resolved the issue by manually reducing the motor offset. Possible that it is something that we can handle on the hardware size, as @BarbourSmith mentioned.

closed 2020-01-26 by amantalion