WebControlCNC/WebControl #85: Pause button stuck in Resume state

gb0101010101, 2020-01-24:
_Pause_ button stuck on _Resume_ when clicked.

## Reproduce
1. Open gcode file
2. Click on _Play_ button
3. Click on _Pause_ button
4. Click on _Resume_ button
5. Button text stuck on _Resume_.

madgrizzle, 2020-01-24:
On which version/branch do you see this happening?  Is it just that the text doesn't change but it does actually resume the cut, or does it not resume?

gb0101010101, 2020-01-24:
It's on _master_ branch pulled today. The button text does not change and the button does not function.

madgrizzle, 2020-01-24:
I can't duplicate the problem. I'm not having issues with pause/resume on the current master branch.

madgrizzle, 2020-01-24:
If one of the threads crash in webcontrol, its possible for it to look like everything is fine on the browser, but it might not respond to commands... You'll see it error out in the terminal screen though.

gb0101010101, 2020-01-25:
Yeah, don't know what happened there. I double checked that I was on clean master and pulled it again. Switched between branches and it was happening on all. Cleared browser caches.

But its not there anymore and I can't reproduce it. Sorry.

closed 2020-01-25 by gb0101010101