WebControlCNC/WebControl #87: No preview of path to cut .nc files

PederSmart, 2020-01-25:
First of all, I am very impressed with this WebControl!!

I'm on the latest version 0920 on a RPI3 and everything is working (love the camera function)
but compare to the old GroundControl I don't see the preview of what my Maslow is about to cut on the board when I open .nc files.
Could be that I am doing something wrong.

Anyway Great project!!
[image]

madgrizzle, 2020-01-25:
It doesn't look like you have a gcode file loaded.  Maybe something is wrong with the NC file that causes WebControl to error out.  Can you email it to madgrizzlemaslow at gmail dot com?

madgrizzle, 2020-01-25:
The file you sent me didn't have the nc extension, but it doesn't really matter..

I am able to open the file fine and preview it.

[image]

I assumed you uploaded the file, correct (GCode->Upload)?  If you could, look at the terminal screen the python app is running in and see if there are any error messages that pop up after you tell it to upload.

PederSmart, 2020-01-25:
I apologize for creating this case. It is working now for me also. You may close this case.  

Again a Great prosject!!!

closed 2020-01-25 by PederSmart

madgrizzle, 2020-01-25:
No problem..