WebControlCNC/WebControl #89: Z axis tabs cause distortions in the GCode cut pattern

amantalion, 2020-01-26:
Hi,

I have a problem with Z axis tabs when I load them in WebControl: Z axis tabs cause distortion in the cut pattern in WebControl. When I run Maslow, then I believe distortions are also present in the contour of the final part. The same GCode is loaded properly in GroundControl. Please, see screenshots below to understand what I mean (distortions are within green circles on the WebControl screenshot).

Tabs in the CAD model:

[image]

WebControl:

[image]

GroundControl:

[image]

Final shape:
[image]

Has anyone encountered the same issue before?

madgrizzle, 2020-01-26:
Webcontrol uses a 3d display of the gcode.  What you are seeing is the perspective of the view.   I'm fairly confident that any issues with the cut are not related to webcontrol.  Note that it looks like you are trying to make a full depth cut.. I think that you might be pushing the capability of the Maslow.  Most people would run multiple passes to get all the way through.

madgrizzle, 2020-01-26:
I rotated the display so you can see.

[image]

It looks like you are cutting down 14 mm (little over 1/2-inch).  That's very deep and it would explain the issues on the right side of the cut.  Rule of thumb is each pass should use a depth of cut equal to 50% the width of the router bit (it's just a rule of thumb.. I tend to not be so conservative and do maybe 80% the width of the bit).

amantalion, 2020-01-27:
@madgrizzle Thank you! Having multiple passes resolved it: I used 80% of the end mill diameter for depth (5mm depth for the 0.25" bit).

However, I've encountered another problem: the width of the square is off by 1/16" (3.9375", instead of 4"). The diameter of the circle is off by ~1/30" (3.5" instead 3.465"). Have you seen this issue before by any chance? Do you think it's a standard error for Maslow?

I have done the Holey calibration 3 times to make sure that it's not due to the calibration.

madgrizzle, 2020-01-27:
That's actually pretty good performance for a Maslow.  Are width and height measurements the same?

amantalion, 2020-01-27:
Yes, width and height measurements are the same for the square.

madgrizzle, 2020-01-27:
Then I think you're pretty well calibrated.  I'd suggest that if you need more accuracy, to tweak your CAD design (make the shapes larger or smaller as needed to get things to come out precisely as you want them).. but 1/16-inch is good.. and 1/30-inch is really good.

closed 2020-01-29 by amantalion