WebControlCNC/WebControl #91: Quick Select Default Ring setting is invalid

idougal, 2020-01-27:
If you open the Quick Configure Menu and select Ring from the pulldow list:

The Rotational Radius turns to 138.4

This number will be described as invalid when you press submit.  It then states that valid numbers are 138.1 or 139.1.

The Advanced Setting rotational radius setting states that 140 should be the default value.

madgrizzle, 2020-01-27:
Thanks.. appears most entries are messed up.  I had to tried to make it so that when using a mobile, the numeric keyboard popped-up.  @gb0101010101 made a change to dist to move on the front page and I'll duplicate that for all the other numeric entries.

gb0101010101, 2020-01-28:
This looks like the same input validation problem I mentioned in another issue. Decimal input precision should probably defined globally and enforced on all inputs. Most of the inputs are limited to 3 decimal points but noticed that UI Unit switch code returns 4 decimals which will cause errors.

madgrizzle, 2020-01-28:
I changed everything to 5 (need to verify disttomove was changed) and pr'ed test1 into master to prepare for a new release.