WebControlCNC/WebControl #93: Best way to save state, if my Pi reboots?

zgerrard, 2020-01-29:
Hi, what's the best way to save state when running GCode? Sometimes my Raspberry Pi reboots or someone unplugs it accidentally and it is very hard to resume again.

Do you usually write down the home coordinates and input those coordinates into settings on the next boot? One thing that I noticed is that on the next boot my coordinate system moves (also, sometimes coordinates of the sled change) and the GCode that I import after I reboot becomes misaligned (and I have to calibrate again, if the coordinates of the sled change).

Thank you for your help!

madgrizzle, 2020-01-29:
The settings (including home coordinates) should persist during a reboot.  if they are not, then something is wrong.  How do you start webcontrol once it reboots?

A bigger problem, however, is that if your RPI reboots during running a gcode, then the controller gets reset.  The controller only writes the motor positions at the end of running gcode to EEPROM to keep it from getting worn out.  If it looses connection with the RPI, then it resets and the motor positions are lost.  Then you'll have to go through the process of resetting the chain lengths (you don't have to recalibrate).

madgrizzle, 2020-02-18:
Closing due to lack of activity.

closed 2020-02-18 by madgrizzle