WebControlCNC/WebControl #102: M0 or M1 stopps Maslow, but no possibility to resume in UI

baxerus, 2020-02-14:
I'm using **WebControl Version v0.922** on a freshly installed Raspberry Pi.

If the Gcode inculed a `M0` Maslow stopps (like it should), but this isn't indicated in the UI (like the Pause button changing to "Resume"). It is actually not possible to leave this state and resume the cut (at least I wasn't able to do so with the buttons in the UI).

I also tried replacing `M0` with `M1` with exactly the same outcome.

May be related to #52

madgrizzle, 2020-02-14:
You hit pause after it paused?  If so, did it change to resume, if so, did it resume after hitting resume?

I have a fix in mind, but curious if that works..

baxerus, 2020-02-14:
Yes, I tried clicking "Pause" after Maslow paused (stood still). The button changed to "Resume". Another click changed it back to "Pause", but still the cut didn't start again.

I also tried the "Play" button. And even the "Stop" button. But the cut never started again. When I remember correctly I actually had to restart the whole WebContro!!

madgrizzle, 2020-02-15:
Ok, I've got a fix for it (seems to work).. it will be in next release

baxerus, 2020-02-17:
If I find the time, I will perhaps test this even before next release.

BTW, is there a documentation on how to do release updates without configuration lost (and perhaps even for non-release updates 😉 )?

madgrizzle, 2020-02-17:
Nothing gets lost during a release update.. it's done within the program (help->upgrade iirc).. but making changes outside a release requires you to download the master branch from the GitHub repo and run it as a normal python program.. after you install all the python requirements..  I'll see if I can do a release today with the fix.

baxerus, 2020-02-17:
Keep you normal release cycle, do not rush to make a release for me!
I will figure out how do do it between releases, *if* I really find time to test in the first place...

baxerus, 2020-02-20:
Testing with version 0.925.

The button is now switching from "Pause" to "Resume" for M0 (I didn't test with M1) and it is also possible to resume a cut. Also the status of the button is communicated between parallel open browser instances. That's all good 👍 

**But** (I'm sorry, but you must have guessed that a "but" will follow 😉) the state isn't "stable". Stable in a sense, that if you do a page reload while button is on "Resume" (or the second browser wasn't online while going into "Resume" button state, like with a smartphone browser) the button is back in "Pause".

It is *not* possible to hit the "Pause" button twice (I tried this on the smartphone). Simply nothing happens.

My suggestion would be, to have the button status be based on the "State" of the job. Since I noticed, that this state is also right after a reload. Or there are easier ways, I do not know...

madgrizzle, 2020-02-20:
Thanks for the catch.  When the browser does a refresh, it requests some settings from webcontrol (units, distance to move) but the state of the pause/resume button isn't currently one of them.  I've got a fixed worked out and it'll be in the next release.  There's a lot of rework that would be nice to do, but right now I'm just patching things to make them work.

baxerus, 2020-02-20:
I totally understand.

I may be able to help from time to time with patches myself, but in the moment time constrains only let me cut stuff now and than and no free time left for software development...

But I can promise to add issues to Github here. In the best documented, tested and researched way possible (I always appreciate good tickets since they help, while bad tickets often only suck 😊)

madgrizzle, 2020-02-20:
That truly helps immensely..

madgrizzle, 2020-02-21:
Try the new release and let me know if it works.  If it doesn't, try Shift-F5 to reload the javascript.  I *think* I have it set to force reload the javascript, but not sure if it's working as intended.

https://github.com/madgrizzle/WebControl/releases/tag/v0.926

baxerus, 2020-03-03:
It's totally working great!

Closing this issue

closed 2020-03-03 by baxerus