WebControlCNC/WebControl #109: Feature request: OrthographicCamera as option in settings

baxerus, 2020-03-03:
I've discovered that there is a [commented out OrthographicCamera in the code of the gcode view] (https://github.com/madgrizzle/WebControl/blob/f6812e867dc7d75c25a38dc7001174dbc2d1362a/static/scripts/frontpage3d.js#L23). It would be great to have an option (in the settings somewhere) to switch to orthographic, because at least for me, the distortion the perspective produces especially on cut lines that are actually on top of each other (like multiple step cuts) is irritating.

madgrizzle, 2020-03-03:
Yeah, I just haven't figured out how to get the cursor position from an orthographic perspective in the cursorPosition() function so that's why its not used.  If someone can figure that out, I can put in an option for changing to orthographic perspective.

madgrizzle, 2020-03-05:
Well, I hacked in a set of parameters that seems to work.  I just ended up trying different values until it came out correct (or close to correct).  I'll work on making this an option now.

baxerus, 2020-03-05:
Super! 👍 

But I now feel a little bit guilty to ask for feature requests and you rushing to fulfill them in less than two days 😄

madgrizzle, 2020-03-05:
Lol.. just had to sleep on it a couple of nights to figure it out what I was doing wrong.  It's not perfect by any means since some values are trial-and-error, but it works.    I also prefer the orthographic view so I made it the default.  In doing all of this, I figured out a bug with the mobile display and fixed it.  So overall, a nice improvement.  This will be in the next release.

baxerus, 2020-03-05:
Nothing said about not being guilty 😄. Or not adding more feature requests 😜.

Will definitely try it out in the next release

closed 2020-03-05 by baxerus

madgrizzle, 2020-03-05:
lol.. shouldn't feel guilty and I like feature requests.  Sometimes, like this one, it prods me to go figure something out I gave up on.