WebControlCNC/WebControl #113: Pre-Release v0.928

baxerus, 2020-03-18:
Cutting path isn't shown anymore after file is loaded.

I tried several reloads of gcode files. Also one that were shown in the past. Changed view to perspective and back (btw otho is cool. At least I think, because no path shown 😉), tried enabling and disabling board view. Nothing worked.

Rolling back to v0.925. Now the path is shown again.

madgrizzle, 2020-03-18:
I'm wondering if the problem is that the JavaScript is still cached in the browser.. I might have forgotten to force a reload.

madgrizzle, 2020-03-18:
For anyone experiencing this, try to force refresh. I think Ctrl-F5

baxerus, 2020-03-18:
I tried hard refresh of the browser (developer console > holding reload button > Hard reload). Problem was still existing.

BTW same on mobile safari, no path shown. Even after (multiple) reload(s).

madgrizzle, 2020-03-18:
Odd.. i wonder if there something I didn't account for that's in you gcode.  Can you email it to me at madgrizzlemaslow at gmail dot com?

baxerus, 2020-03-18:
Exactly the same gcode is working in v0.925 (adding it in here, because its less than 50 lines. And it is than linked to this issue):

```gcode
(1001)
(T12  D=3.175 CR=0 - ZMIN=-19.5 - FLAT END MILL)
(USEACTIVESPINDLE AND STOPSPINDLEFORG0 OPTIONS ACTIVATED)
G0 G40 G90 G17
G21

(DRILL HOLDER SCREWS)
T12 M6
G0 Z3
G0 X267.412 Y14
M3 S15000
G1 Z-19.5 F500.
G1 Z3
M5
G0 X198.588
M3 S15000
G1 Z-19.5 F500.
G1 Z3
M5
G0 X51.5 Y107.412
M3 S15000
G1 Z-19.5 F500.
G1 Z3
M5
G0 Y138.588
M3 S15000
G1 Z-19.5 F500.
G1 Z3
M5
G0 Y207.412
M3 S15000
G1 Z-19.5 F500.
G1 Z3
M5
G0 Y238.588
M3 S15000
G1 Z-19.5 F500.
G1 Z3
M5

G0 Z3
M5
M30
```

madgrizzle, 2020-03-18:
Thanks for the gcode.  I'm sure it's a parsing issue.

madgrizzle, 2020-03-18:
Yes, definitely a parsing issue (I used improperly made gcode to develop it).  So I need a little rework.  I pulled the release.  Sorry for the inconvenience.

baxerus, 2020-03-18:
No inconvenience. Instead thank you for your great work! 👍

madgrizzle, 2020-03-18:
try the new version (0.929)

kick the tires well if you would :)

baxerus, 2020-03-18:
Gcode path are shown now (with v0.929).

Actually not tested anything else, because have to work and just put the software on there in 5min or so between to calls, to not let you hang here.

madgrizzle, 2020-03-18:
Ok, let's leave this open until you get a chance to try it out more.