WebControlCNC/WebControl #120: Feature Request: GCODE "R" command filter

Orob-Maslow, 2020-03-31:
On occasion, I get gcode files with the radius code in them and I have not seen one for some time, but when the R command gets sent to the arduino, the maslow system just freezes.  Can we filter gcode for R and either warn if we see it or delete the R portion, but send the X-Y command?

madgrizzle, 2020-03-31:
Is there a way to convert it to i/j?

gb0101010101, 2020-05-12:
This should really be supported in Maslow Firmware as R is valid for G2/3 commands.
Using R you can work out I & J. You cannot remove R as it is required.

Found this snippet in Marlin firmware which could easily be adapted to work out I J and added to Maslow Firmware G2()
```
      float arc_offset[2] = { 0.0, 0.0 };
      if (code_seen('R')) {
        const float r = code_value_axis_units(X_AXIS),
                    x1 = current_position[X_AXIS], y1 = current_position[Y_AXIS],
                    x2 = destination[X_AXIS], y2 = destination[Y_AXIS];
        if (r && (x2 != x1 || y2 != y1)) {
          const float e = clockwise ? -1 : 1,                     // clockwise -1, counterclockwise 1
                      dx = x2 - x1, dy = y2 - y1,                 // X and Y differences
                      d = HYPOT(dx, dy),                          // Linear distance between the points
                      h = sqrt(sq(r) - sq(d * 0.5)),              // Distance to the arc pivot-point
                      mx = (x1 + x2) * 0.5, my = (y1 + y2) * 0.5, // Point between the two points
                      sx = -dy / d, sy = dx / d,                  // Slope of the perpendicular bisector
                      cx = mx + e * h * sx, cy = my + e * h * sy; // Pivot-point of the arc
          arc_offset[X_AXIS] = cx - x1;
          arc_offset[Y_AXIS] = cy - y1;
        }
      }
      else {
        if (code_seen('I')) arc_offset[X_AXIS] = code_value_axis_units(X_AXIS);
        if (code_seen('J')) arc_offset[Y_AXIS] = code_value_axis_units(Y_AXIS);
      }
```

madgrizzle, 2020-05-12:
I don't have any control over the stock firmware , but we can look at incorporating this into the holey branch I have on my firmware fork.

gb0101010101, 2020-05-12:
@madgrizzle You don't need control of stock firmware. We can submit a regular PR. With a few votes it will get auto committed.

madgrizzle, 2020-05-12:
But never in an official release.  Last release was January 2019.

gb0101010101, 2020-05-14:
Here is a preliminary PR #137 that supports rendering G2/G3 using R in WebControl.

@Orob-Maslow can you provide test gcode file that uses R for both G2 & G3 commands?

@madgrizzle Right, Maslow has not increment firmware version number in a while but its still in github and can be compiled.

Orob-Maslow, 2020-05-14:
I have attached the offending file.  It only has one  G03 for each depth, but should work for testing.  I found it interesting the R's are rarely generated and when they are typically they are for gigantic radius values, so I've been replacing them with straight lines manually.  G03 or G02 -> G01 and just delete the R and its value.  This feature request was to incorporate a similar filter to remove the R value, but I agree this should be done right and incorporating the Marlin code mentioned above is a very good approach.
[home-cutout2_0001 - .gcode.txt](https://github.com/madgrizzle/WebControl/files/4628119/home-cutout2_0001.-.gcode.txt)

I thing we should close this request since #137 is incorporating this feature.  I don't have a file that does both G02 and G03 with currently, but I'll generate one and attach it to #137

gb0101010101, 2020-05-15:
Added Maslow Firmware PR
https://github.com/MaslowCNC/Firmware/pull/541

@Orob-Maslow Please test Maslow Firmware and WebControl changes. Let me know if you have any problems. I do not have access to physical machine right now so really need someone to test this live.

I tried your gcode and it worked but it is not a good test case. Added G3 test code in Firmware PR.