WebControlCNC/WebControl #124: half working after adjusting Z axis

MFD72919, 2020-04-16:
So I am pretty new to this whole thing.

Was having some issues getting my maker made to cut to the proper depth. Prior to clicking and changing the Z axis button everything was working great and i was very pleased.

after clicking that and changing the value i can no longer access any buttons. i can drop down the menu for Gcode but non of the buttons work. i click the action button and nothing happens. i can drop down the settings button but nothing in there works.

I can still move the machine with the arrow keys so its still connected.

any ideas how i can resolve this? is this something i did or a computer issue.... PLEASE HELP

madgrizzle, 2020-04-16:
Which version are you using?  make sure you are using 0.932

MFD72919, 2020-04-16:
.932

i even uninstalled then reinstalled just to be sure. same problem still idk why....

it was running fine all day for me

madgrizzle, 2020-04-16:
hmm.. when you see behave like that, it means some crashed.  can you repeat it and screenshot the messages in the terminal window that's running it?  Is this windows, linux, or raspberry pi version?

MFD72919, 2020-04-16:
it is windows. and i cant do anything with it now in turms of messages. i cant upload gcode. or click any buttons. the only buttons that work are the arrow keys to move the sled and the home button (which doesnt actually send it home but does move the motors)

madgrizzle, 2020-04-16:
can you screenshot the terminal window that its running in?  I want to see if there are any error messages.  When you click it to start, a terminal window opens up and the browser launches.

MFD72919, 2020-04-16:
[image]
[image]

madgrizzle, 2020-04-16:
And when you click Actions, does anything appear on the terminal window?

MFD72919, 2020-04-16:
9cf6895a0634f9db476b5f58b2988

MFD72919, 2020-04-16:
then it just popped up client discounnected. getting photo now

madgrizzle, 2020-04-16:
That's normal.  But the action menu doesn't come up?

MFD72919, 2020-04-16:
[image]

MFD72919, 2020-04-16:
nothing opens. even if i go into settings or gcode the menu drops down but when you click anything nothing happens

madgrizzle, 2020-04-16:
have you tried restarting the browser?  Maybe force refresh (Ctrl-F5) I've not seen this behavior before.  You're running Edge?

MFD72919, 2020-04-16:
yes edge. it was running flawlessly all day was super happy with it. i will try the refresh now

madgrizzle, 2020-04-16:
>after clicking that and changing the value i can no longer access any buttons.

What value did you change?  And to what did you change it to?

MFD72919, 2020-04-16:
no change after refresh except it says not connected but i can still move the sled with the arrows

and z axis said .1 and i was trying to change it to .5.

madgrizzle, 2020-04-16:
scroll up on that terminal screen to where it starts and see if there are any messages at the beginning.. something about error or something.

MFD72919, 2020-04-16:
nothing different than what i sent before

madgrizzle, 2020-04-16:
Try reaching it via 192.168.1.3:5000 instead of localhost:5000.  I'm sort of grasping at straws here.  Have you rebooted the computer?

MFD72919, 2020-04-16:
yep i restarted it after i did a reinstall. i may try and wipe it out again tomorrow and reloading it all over. i just dont know what i did to cause this to happen kind of frustrating because it was working so well... for what its worth the calibration process went crazy smooth

madgrizzle, 2020-04-16:
Hit Ctrl-Shift-I in the browser and it will bring up a debugger screen.  Then click the 'console' tab on the debugger.. you should see a bunch of messages.  Click actions and screenshot it.  This is what mine looks like after pressing Actions
[image]

MFD72919, 2020-04-16:
bunch of errors stand by

MFD72919, 2020-04-16:
[image]
[image]

madgrizzle, 2020-04-16:
Well, that's the problem but I have no clue what's causing it specifically.  I'm THINKING that there is a some setting that is wrong and its having a problem with it.  Can you email me the webcontrol.json file.. send to madgrizzle maslow at gmail dot com.  You can find the file in C:\users\azide\.WebControl

Note there's a period in front of WebControl

madgrizzle, 2020-04-16:
madgrizzlemaslow at gmail dot com

MFD72919, 2020-04-16:
sent

madgrizzle, 2020-04-16:
Ok, I was right... good!  Open that file with a text editor (notepad is fine) and search for the word NaN.  It stands for 'Not A Number' (computer thing).  You should find it under the 'distToMoveZ' entry.  Change that from NaN to 0.5 and save it.

madgrizzle, 2020-04-16:
Something glitched when you changed it.  I can't readily duplicate it, so its hard to tell.

madgrizzle, 2020-04-16:
Its near midnight here.  I'm fairly confident that will fix your problem.. I'll check back in the morning.

MFD72919, 2020-04-16:
it wont open at all now lol dialog box opens then closes... i just deleted it off my harddrive going to redownload....

madgrizzle, 2020-04-16:
Let me send you a corrected file to replace the one.  I'll reply to your email.

madgrizzle, 2020-04-16:
Redownloading won't help.  Try the one I just sent.

MFD72919, 2020-04-16:
all good now the reload. thank you!!

closed 2020-04-16 by MFD72919