WebControlCNC/WebControl #127: Turning Router on/off

ersdds, 2020-04-27:
I was wondering if any of the AUX outputs in Web Control Can Power the Router on/off

Orob-Maslow, 2020-05-10:
Webcontrol is the software that communicates with the arduinto controller.  The aux ports are on the arduino and can be activated by sending gcode such as M3 or M4 for different spin directions and M5 for off according to a github/grbl issue.

gb0101010101, 2020-05-12:
AUX1 on the Malsow controller board is mapped for router/spindle power. Use this to control an optional optically isolated relay.
Maslow firmware only supports router/spindle in one direction, so gcode M3 & M4 do the same thing, and will turn it on. Gcode M5 is off.
WebControl passes M3/4/5 commands to controller.

closed 2020-05-23 by Orob-Maslow

ersdds, 2020-05-23:
Thanks for the clarification