WebControlCNC/WebControl #129: Touch Zero doesn't stop when it touches 0

ersdds, 2020-05-01:
The touch Z axis engages the machine to move down but it doesn't seem to stop when it makes contact with the sensing block. I am connected to AUX4 of the original Maslow motor shield.

madgrizzle, 2020-05-01:
I looked at the firmware and there doesn't appear to be anything that webcontrol needs to do to get it to work if you can get it to start probing.. that is, once the gcode is sent and the machine moves, it's all firmware-related after that.  I'm not really sure what to suggest other than to create a message on the forum.  Perhaps post an image of how you have it wired up as well so people can see it.

davidelang, 2020-05-01:
silly question, you are connecting this pin to ground, not power, right?

David Lang

ersdds, 2020-05-02:
My touch zero is coming off of AUX4 of the motor shield. It is going from the top pin to the bottom pin. It was working under ground control. I will put a meter across the wires going to the router and "grounding plate" to make sure it is still working. Or I can fire up Ground Control again and make sure it's still working

ersdds, 2020-05-02:
Gentleman, I really appreciate all your help. It was my screw up. I replaced the connector onto Aux 1 instead of AUX4. It worked perfectly once I connected it to the right spot.

closed 2020-05-02 by ersdds