WebControlCNC/WebControl #132: add buttons in chain settings to let you move up/down by 1 chain link

davidelang, 2020-05-10:
one thing that it would be good to do is to let people set the position of the chain length reset. you can do this manually in the settings, but it would be nice to have buttons that let you extend/retract both chains by 6.35mm and then save this with a button rather than having to go into settings

note, currently the firmware does not support this fully, because the extension distance is modified by chain sag/stretch, so when you extend to a given length, the sprockets may not quite be at 12 o'clock, but fixing that would require significant firmware changes.

madgrizzle, 2020-05-10:
I don't think chain sag/elasticity comes into play with the extend chain as it uses singleaxismove, which doesn't call any of the kinematics.  The idea make sense and I agree that setting really should be multiples of chain pitch rather than mm.  But we can make webcontrol do all the conversions on that page.  Would be a good time to add in the automatic calculation of the number of links and user can adjust from there based upon their preference.  Some might like it higher on the board and others lower.  I think the number of links that put the sled closest to the center should be the default.

Orob-Maslow, 2020-05-15:
so would you use the quick configure measurements to determine where the center is and then put the sled closest within an integer multiple of chain links muliplied by the chain pitch?

davidelang, 2020-05-15:
I would use the normal calibration (ideally holey) and then let the user move 
up/down by one-link increments until the user is where they want 'home' to be 
and then store that length and tell the user to mark the links currently at 12 
o'clock

David Lang

madgrizzle, 2020-05-15:
> so would you use the quick configure measurements to determine where the center is and then put the sled closest within an integer multiple of chain links muliplied by the chain pitch?

Yes, and then as @davidelang suggested, let the user move it up/down a link.  It would be nice if there was a button that moved both chains but I don't know if that the firmware can handle simultaneous "singleaxismove" commands.  I think I tried at one time and it didn't work out as expected.

davidelang, 2020-05-15:
currently it can't, so the one button would have to drop one side, then the 
other side.

David Lang

On Fri, 15 May 2020, madgrizzle wrote:
>
> Yes, and then as @davidelang suggested, let the user move it up/down a link.  It would be nice if there was a button that moved both chains but I don't know if that the firmware can handle simultaneous "singleaxismove" commands.  I think I tried at one time and it didn't work out as expected.
>
>