WebControlCNC/WebControl #135: Alert     Error with returning to home.

PederSmart, 2020-05-10:
Thank again for a great solution to control my MaslowCNC.
Everything is working fine, except every time i press the Home button I Get this  pop-up    

«Alert     Error with returning to home.»

I don’t see any error in the Controller Messages window
I have download the diagnostics file, but when I search in it, there is nothing that match this alert.

I’m running on an RPI 3 (updated) and WebControl version 0.9333

I the same message on Stock Firmware:1.26 or Holey Firmware:51.27

I have tried all three «Wipe Controller ...»

The same alert on the original MaslowCNC shield v1.1 and from MakermadeCNC Shield v1.2

Any suggestion on how to troubleshoot this problem?

madgrizzle, 2020-05-10:
Can you tell me what the home coordinate value is?  Its reported on the main screen under the arrows (can also find it in settings)

madgrizzle, 2020-05-10:
And the z axis safe height.. it's found in Maslow Settings.

PederSmart, 2020-05-11:
Thank you @madgrizzle for guide me in the right direction.
The Maslow Settings: Z-axis Safe Travel Height in MM, was 0,01

Changed it to default setting: 5

And Home-button is back and working.

Thanks again

Will this solution be found, if someone else is getting this alert and tries to «Google» this Alert?

I found nothing when I tried .....

madgrizzle, 2020-05-11:
It's the comma causing the issue.

PederSmart, 2020-05-11:
I changed It now to «2» and Home-button is still working. So you are probably right

madgrizzle, 2020-05-11:
This issue came up before (commas instead of periods for decimal points).  @gb0101010101 was/is looking at it, but we haven't "fixed" it yet.  Maybe we'll add a note somewhere about commas in the fields until we can accommodate them.

PederSmart, 2020-05-11:
I’m using metric units

closed 2020-05-11 by PederSmart

gb0101010101, 2020-05-12:
PR #130 fixes this for all settings. The decimal inputs now support comma as separator and are correctly converted to decimal float. Should work for all OS languages as its handled natively by browser.