WebControlCNC/WebControl #136: Feature Request: Z axis mechanism safety limits

ariasr, 2020-05-14:
In order to prevent damages to the z axis mechanism and motors, it would be useful to be able to set upper and lower limits for the movement of the z axis mecanism that are independent of the drill bit set zero calibration. This idea is similar to the working area limits where the drill bit never goes outside the working area.

ariasr, 2020-05-14:
[image]

It could be as simple as adding a Set Upper Limit, Set Lower Limit buttons, with similar behavior to the Set Zero button, and just check for the limits when moving the z axis.

davidelang, 2020-05-14:
how do you know where you are to set the limits?

gb0101010101, 2020-05-14:
Yep, I've been think about this too. I think it should be part of the calibration process as it really should not change (whereas zero point can change every time you change router bit).

During calibration:
- Provide toggle to turn zAxis on/off.
- If On, then provide decimal input for distance and raise/lower buttons.
- Provide "Set top" and "Set bottom" buttons.
- Save as two new settings.
- Incorporate into gcode file load process to make sure all code stays with limits and warn if not.

madgrizzle, 2020-05-14:
I think this really should be a firmware thing.  If you add a variable in the firmware for bit length and that's what changes when you define zero, then you can leave the steps value alone (just subtract the bit length from the steps value) and set a min/max for steps for limiting.

ariasr, 2020-05-14:
> how do you know where you are to set the limits?

Similarly to how you set Zero. You could move the Z axis up and down while looking at the mechanism, when you reach a point where you feel conformable with hit the "set Upper/Lower Limit buttons" to record the position.

ariasr, 2020-05-14:
> I think this really should be a firmware thing. If you add a variable in the firmware for bit length and that's what changes when you define zero, then you can leave the steps value alone (just subtract the bit length from the steps value) and set a min/max for steps for limiting.

You could either check on webcontrol for any gcode command that will make you exceed those limits and send a warning before sending the command to the controller, or put the check in the controller to not move the z axis past the limits.

I see there is active development in webcontrol, don't know how much for the firmware, last update was a while ago.

madgrizzle, 2020-05-14:
Is this with no bit in the router?

ariasr, 2020-05-14:
> Is this with no bit in the router?

Yes, without the bit for the lower limit, juts to set the physical limits of the z-axis

madgrizzle, 2020-05-14:
The problem comes in when you install the router but and define zero.  It sets the steps (i.e. height) variable in the firmware of the z axis to zero.  The upper and lower limits are then referenced to something that is no longer tracked.  So you would either have to add additional variables into the firmware (as I suggested) or do all of it in webcontrol. The problem in doing the latter is when webcontrol and firmware get out of sync for some reason...  I really think the firmware solution is far superior (and simpler) way of accomplishing it.

ariasr, 2020-05-14:
> Yep, I've been think about this too. I think it should be part of the calibration process as it really should not change (whereas zero point can change every time you change router bit).
>
> During [...]

Depending of the setup. If you never take the router out, it makes sense for it to be on the calibration.

However if you do take the router out, you might want to set those limits more often. Because the router position might affect your lower limit, as the z axis mechanism could go further but the router could hit the sled.

The upper and lower limits could be because you reach the end of the threads of the screw of because the router is touching the sled.

ariasr, 2020-05-14:
> The problem comes in when you install the router but and define zero. It sets the steps (i.e. height) variable in the firmware of the z axis to zero. The upper and lower limits are then referenced [...]

Its more elegant for sure. Are you able to modify the firmware?

madgrizzle, 2020-05-14:
It's certainly possible to do so.  I think it's better to be done by someone that can readily test and debug it.. they'd have to have a working zaxis to check it out.

ariasr, 2020-05-14:
I have a Z axis to test it on, but I have 0 knowledge on Arduino programing and the Maslow architecture or protocols to communicate back and forth.

madgrizzle, 2020-05-14:
I hear you..I have too much on my plate to try to tackle it.  I don't think it would be very hard do, but you'd need a setup to really test it (,can't rely upon fake servo mode)

ariasr, 2020-05-14:
I am going to give it a shot at Arduino programing. Keep you posted on how it goes!

Orob-Maslow, 2020-05-14:
I'll test when you are ready if you like.  Let me know.

ariasr, 2020-05-14:
Awesome!

madgrizzle, 2020-05-14:
>I am going to give it a shot at Arduino programing. Keep you posted on how it goes!

My recommendation is to do practically everything in the Axis class.  Create a variable called axisOffset that defaults to 0 and adjust all the calls so that it is part of the calculation and returned values.   The G10 function in gcode.cpp needs to me modified to set the axisOffset value when it receives the command to zero the Z Axis.. something like zAxis.setOffset(zgoto) instead of zAxis.set(zgoto).

Finally, the axisOffset values need to be stored in EEPROM.  It would make sense to create a settingsStepsV2 structure that contains each axes' offset.

Just work on it and we'll help you get through it :)

davidelang, 2020-05-15:
the protocol to send back and forth is ascii, this would be additional B 
commands.

the firmware is programmed in C, and the code is fairly easy to understand.

David Lang

On Thu, 14 May 2020, ariasr wrote:

> I have a Z axis to test it on, but I have 0 knowledge on Arduino programing and the Maslow architecture or protocols to communicate back and forth.

ariasr, 2020-05-15:
> the protocol to send back and forth is ascii, this would be additional B commands. the firmware is programmed in C, and the code is fairly easy to understand. David Lang
> […](#)

are the B commands specific to Maslow, and can be added as need be without complying to a predefined standard set of commands?

madgrizzle, 2020-05-15:
I would repurpose the G10 command, not create a new B command

ariasr, 2020-05-15:
From Wikipedia "G10 | Programmable data input | M | T | Modifies the value of work coordinate and tool offsets" Looks like that's the one.

davidelang, 2020-05-15:
they are specific to the maslow.

David Lang

davidelang, 2020-05-15:
that could be appropriate for setting the Z=0 when you change the tool, but not 
for defining what the limits are in the first place.

David Lang

On Fri, 15 May 2020, ariasr wrote:

> From Wikipedia "G10 | Programmable data input | M | T | Modifies the value of 
> work coordinate and tool offsets" Looks like that's the one.

ariasr, 2020-05-15:
This is the current implementation of G10. It expects a new value to reset Z. Like G10 Z 0.0 to zero Z axis. 

How about?

G10 Z0 //grap current position and set to lower Z limits
G10 Z1 //grab current position and set to upper Z limit

void  G10(const String& readString){
    /*The G10() function handles the G10 gcode which re-zeros one or all of the machine's axes.*/
    float currentZPos = zAxis.read();
    float zgoto      = sys.inchesToMMConversion*extractGcodeValue(readString, 'Z', currentZPos/sys.inchesToMMConversion);

    zAxis.set(zgoto);
    zAxis.endMove(zgoto);
    zAxis.attach();
}

ariasr, 2020-05-17:
I have got the firmware ready for testing. I added the following commands.

B17 : Set the upper limit to the current Z axis position
B18 : Set the lower limit to the current Z axis position
B19 : Clears the upper and lower limits
B20: Echoes back the upper and lower limits stablished

The next step is to add that functionality to the Z axis screen.

I am new to GitHub and collaborative development. Whats the protocol... should I do a Pull Request at the Firmware Site, and a Pull request at webcontrol site?

davidelang, 2020-05-17:
pull request at the firmware site for the firmware changes
pull request at the WC site for the WC changes

please do these changes for the stock and the holey firmware both if possible 
(until we can all agree to abandon the original calibration, or find a way to 
merge them so either can be used)

David Lang

ariasr, 2020-05-17:
OK

Orob-Maslow, 2020-05-17:
The B commands in gcode are reserved for rotational movements around the Y axis for multi axis cnc's (according to (https://en.wikipedia.org/wiki/G-code, but https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands is different).  Perhaps the Z axis limits should be a G command subset as @madgrizzle suggested earlier, or maybe go with GRBL code commands which use $ to set details.   I guess it is a matter of which "standard" we seek to adhere to.  

The testing approach will be to flash the current branch on a fake servo simulated maslow with no motors to see if it responds correctly and then put it on a real system.

ariasr, 2020-05-17:
I agree we should stick to the standard.

So the dilema is where to put the commands. 

I dove in the task with the intend to use G10, but after doing some reading I concluded that G10 has a standarized functionality and parameters it should take, that are different from what we are looking for.

After digging in the Maslow code I noticed there are 16 B codes currently in use on Maslow that are used for machine specific configurations. So I leaned towards that defecto Maslow way of doing things.

At the the end its just a mater of form.

The good news is that the functionality is working great. Limits are easily set up, and coexist perfectly with the Z touch calibration, which I just tested and loving it.

Here is a picture of my pride and joy z axis sled that I am working on.

[image]

Orob-Maslow, 2020-05-17:
That sled looks awesome!  You need to post that pic on the maslow forum because I'm sure a lot of people would like to see it.  I have some questions about it as well but I would rather stick to the code issue at hand on this thread.

gb0101010101, 2020-05-18:
@ariasr You can use M commands that are typically used as setup commands or for custom functions. Anything >= M200 should be safe.

BTW nice sled  - did you cut that on Maslow?

davidelang, 2020-05-18:
in that case, we should change GC/WC/firmare to use something other than the B 
prefix

in the maslow world, B## codes are for maslow specific commands that would not 
make sense for a normal CNC

these are things that are pretty much unique to the maslow kinematics, so there 
is nothing close in the standard gcodes.

the grble $ settings are not good matches, they just change settings, where the 
maslow B codes frequently move the machine.

any other thoughts on how to handle this?

David Lang

On Sun, 17 May 2020, Orob-Maslow wrote:

> The B commands in gcode are reserved for rotational movements around the Y
> axis for multi axis cnc's (according to (https://en.wikipedia.org/wiki/G-code,
> but [...]

ariasr, 2020-05-18:
> @ariasr You can use M commands that are typically used as setup commands or for custom functions. Anything >= M200 should be safe.
> 
> BTW nice sled - did you cut that on Maslow?

Thank you.... 

I had it laser cut, just because my standard z axis was broken a that time. It should work fine with the Maslow. I have cut acrylic before with Maslow and it works fine.

ariasr, 2020-05-18:
I will go ahead and do a pull request in the firmware repository. They likely have something to say about which code to use. After all they are the guardians of the firmware. Once that is settled, ill go ahead and implement the calls to the firmware in web control.

Thanks a lot for your input and encouragement... less than a week ago I was very shy about Arduino and Maslow code. Today I have a new skill in the tool bag.

Cheers

Cheers

ariasr, 2020-05-18:
Firmware pull request uploaded

Orob-Maslow, 2020-05-19:
I see the pull request for the firmware.  I'll compile and hopefully flash tomorrow.  I was messing with webcontrol and came up with this:

[image]

I am trying to remember how to get the action from the button to translate to an action in actions.py.  No data other than the command needs to pass to the arduino because the information is in the firmware on the arduino, so it just needs the B## code sent over serial.

Orob-Maslow, 2020-05-19:
Oh and quick question.  for those of us with z axes with belts, the motors run in reverse.  does this work with a + and - zaxis pitch?

ariasr, 2020-05-19:
Yes, Y have the same setup. And it worked great.

ariasr, 2020-05-19:
Thats exactly what I had in mind. Its a really easy upgrade. I was planning on coding it myself, but you already got a head start. Do you want to finish it? Whats left to be done is just the code for the actions, which are basically cut and paste and a little editing.

ariasr, 2020-05-19:
> Oh and quick question. for those of us with z axes with belts, the motors run in reverse. does this work with a + and - zaxis pitch?

Yes, I have the same setup. And it worked great.

ariasr, 2020-05-19:
> I see the pull request for the firmware. I'll compile and hopefully flash tomorrow. I was messing with webcontrol and came up with this:
> 
> [image]
> 
> Here it is sending the commands with the button presses:
> [image]

Thats exactly what I had in mind. Make a pull request so that we can test it.

gb0101010101, 2020-05-19:
I don't think these buttons should be in the z Axis modal. I doubt people will be constantly setting Max and Min as you only need to change it when you remove the motor. I think its more apt to make it a calibration step.

Orob-Maslow, 2020-05-19:
After thinking about it, I agree with that.  They should be in quick configure or another configuration area so they don't get pressed by fat fingers on a phone easily.  

Having said that, they sure worked nice to test and I have an update.  After flashing the firmware and running the version of webcontrol shown with the "non-ideal" buttons, I can say that it works.  I took video, but it is probably easier to read:

The firmware code is as stated.  B17 is max, B18 is min, B19 clears the value.  I moved the z axis, set the lower limit
[image]
then moved the axis and set the upper limit
[image]
then tried to move the z axis above and below that limit.  It goes to the limit and stops.  I would like a log error message or indication that passing the upper/lower limit was attempted so I know where to look for issues.  After moving to each limit and trying to pass it, I reset them with the third command.
[image]
 and was able to move beyond the previously imposed limit as expected.  I don't have a button to check or location to display the values for the actual settings, so that fourth command did not get tested initially.  I changed one of the macro buttons to send B20 and it works, but I had cleared the results, so nothing returned the first time.  I then set the high and low to the current value and the B20 command returned the same value for them both.  
[image]
I call this a success with the suggestion that there be a non-blocking warning or error passed for the user to know a potential damage attempt was prevented and that we get further agreement on where the zaxis limit buttons should be placed.