WebControlCNC/WebControl #139: Discussion: Release process

emilecantin, 2020-05-18:
I'd like to start a discussion regarding the release process. I know we currently use the even=stable, odd=experimental convention, but is there a reason why all of them are marked as "pre-release"?

I think it'd be way easier to actually use that pre-release feature for experimental releases. That way, we could promote experimental features to stable once they've been tested; no need to cut an additional release.

It'd also make it easier to consume the list of releases from a script. I'll probably need this as I improve on the RPI image build.

Orob-Maslow, 2020-05-18:
Perhaps it is a matter of definition.  Only at 0.932 did the even odd thing
start.   I understand release to mean good for the general population while
prerelease is not fully vetted... However rigorous that is.  What would be
a better description that should be used?

At what point does it move to 1.0?

emilecantin, 2020-05-18:
No, I think that description is fine. It's just that when you go on the Releases page, everything there is labelled "pre-release", which is the default value for that flag. I think we should make use of that flag, instead of layering our own convention on top of it.

Orob-Maslow, 2020-05-19:
Agree.  Even or odd, if stable, it should be a release.  If testing new or specialized features, then prerelease... is there also a major/minor release approach?

madgrizzle, 2020-05-19:
I'm open to any option that works and prefer the "correct" way, whatever that is.

It was easy to implement the even-odd thing to solve the problem of releasing a release that breaks webcontrol.. including upgrade/downgrade.  I wanted to make it so people could opt-in to testing and those that just want stability, don't even see releases that are not "tested".  

As for the pre-release tag, I just never used it.. nothing against it.

emilecantin, 2020-05-19:
As of right now, all the releases are marked as "pre-release". I think it wouldn't be too hard to go through them and remove the "pre-release" flag on the stable ones, as a first step.

The "isExperimental" method in the update code could then be updated to look at the [`prerelease` flag](https://pygithub.readthedocs.io/en/latest/github_objects/GitRelease.html#github.GitRelease.GitRelease.prerelease) instead of parsing the version number, and that would bring us to the exact behaviour we have today.

The main differences would be that it'd then be possible to "promote" a release as stable (or not, no obligation), and it'd be a bit clearer for people browsing the releases page what's what.

madgrizzle, 2020-05-19:
Is there any real reason to keep all the old releases?

emilecantin, 2020-05-19:
Not many. You mentioned the downgrade possibility earlier, and it could also help pinpoint when a possible regression was introduced.

Having the history that goes with the releases (the description of the changes on each release) is very valuable, though. I refer to it all the time in other open-source projects.

I don't think it's a significant cost to keep them, so I don't really see a great reason to get rid of them.

madgrizzle, 2020-05-19:
I was just thinking of ease of moving forward using pre-release tag, but I'm sure we could just add another conditional to make sure the version number is greater than 0.9335 (or whatever) before tagging it in the release manager as an "experimental".

emilecantin, 2020-05-19:
Not sure why you'd need to do that. What I had in mind was pre-release === experimental. No need to parse the version number or anything. The only prerequisite would be to manually go into the old releases and correct the flag for all of them. There's only 18 releases right now; shouldn't take very long. I can do it myself if you give me access.

madgrizzle, 2020-05-19:
I went ahead and fixed it.

Regardless, you are a member of the organization, but don't have access to the WebControl repo, correct?  When I brought the repo over, it brought over all the collaborators as well (which you weren't one of, I guess).  Do I need to specifically add you to the WebControl repo?  Or do we need to create a team or something for the organization?  I'm new to all this.

emilecantin, 2020-05-19:
Awesome, thank you.

And yes, that's right; I only have read access to the repo, and no, I wasn't a collaborator before (just now getting involved in this). I think creating a team would be the best approach for this.

madgrizzle, 2020-05-21:
I'm creating a final release on the madgrizzle repo marked as stable (0.94, even digit).  I did a PR to pull in all changes from my fork into this repo so they should be identical.  I'd like to get the same release up on this repo as well and will work on modifying my build scripts to build and publish here.  However, I'd like to discuss if there are alternative ways to do the builds.

Right now I build for RPI, linux, Win32 and Win64.  The RPI build runs on a dedicated RPI.  The Win64 build runs on a win64 computer and the linux and Win32 builds run in VMs on that same win64 computer.  The linux machine builds the firmwares and then makes them available on its ftp server for the RPI, Win32, and Win64 machines to download them (so they don't have to build the firmwares).  I don't launch the RPI, Win32, or Win64 build scripts until the linux build script has finished with the firmware compilations.  After all the builds are completed, each machine uploads them via FTP to the linux machine's "release" directory.  On the linux machine, I then run a script that creates a github release and uploads the builds.  Finally, I edit the release via web to add a title and description.

Any idea on ways to improve this or maybe, possibly, put it all in the cloud?  The builds are pyinstaller based.  I'm not sure if there's a way to cross-compile, etc. so it runs on just one machine.  Or do we need to maintain VMs and if so, is there a VM for an RPI?

emilecantin, 2020-05-22:
I think the first step here would be to actually publish these scripts, so people could build releases on their own.

Then, we could look at integrating a continuous integration tool (like TravisCI); there are a handful out there that are free for open-source projects. This would be a "cloud build", in your words. They're usually able to build in a few different environments, Travis even has macOS environments available.

I'm not really familiar with pyinstaller, so I'm not sure what it actually needs to build natively; unless you wrote some native code yourself as part of WebControl, we can probably get away with just downloading pre-built versions of every dependency.

Compiling software for the Raspberry Pi is something that's pretty common; depending on our specific needs (again, need to research pyinstaller) we may have many options here. Worst case scenario would be to move all the dependency management to the pre-built image, might make things a lot simpler from the WebControl side.

For the firmware build, I think that should actually live under the firmware's repos; the release script would only download the latest release binary from the firmware repo directly. That might not be possible, however, as we might not have control over all versions (notably the stock one); in these cases I think we should either maintain a fork that has the CI integration, or put in a pull request for that to Bar so we can pull from the "true" source.

madgrizzle, 2020-05-22:
>I think the first step here would be to actually publish these scripts, so people could build releases on their own.

The .spec files to build the pyinstaller releases are already part of the repo.  main.spec is the spec file for the singlefile variant and main-onedir.spec is for the singledirectory variant.  I use the same spec files for each architecture and it seems to ignore the architecture-specific adds in them.  The user will have to adjust the pathex within the file to make it work on their installation, but that should be simple enough to do.

What I haven't pushed is the scripts I used to semi-automate the process.  That's because its specific to how my build environment is and secondary that it includes things like FTP passwords and such.

>For the firmware build, I think that should actually live under the firmware's repos; the release script would only download the latest release binary from the firmware repo directly. 

This is the challenge because the last release was nearly a year and a half ago.  Only bar can make the releases, afaik.

>in these cases I think we should either maintain a fork that has the CI integration, or put in a pull request for that to Bar so we can pull from the "true" source.

I really would like to see a single firmware that has both triangular calibration (i.e, stock) and holey calibration contained in one.  If we do this, then I think we can convince bar to make it the official release if we demonstrate that it also works with ground control, which knows nothing of holey calibration.  It's not impossible, or even hard, but its not trivial and will take time and testing.

madgrizzle, 2020-05-22:
>I'm not really familiar with pyinstaller, so I'm not sure what it actually needs to build natively; unless you wrote some native code yourself as part of WebControl, we can probably get away with just downloading pre-built versions of every dependency.

Afaik, you need a complete installation with all dependents installed on each architecture, to the extent that the program can actually run, in order to use pyinstaller to compile the program.  I'm not sure how all this can be done in the cloud.  Travis-CI is completely new to me.  I understand what it does, but how to set it up is way beyond me.

madgrizzle, 2020-05-22:
And so it dawned on me that I didn't create a -dev branch to put the recent changes into.  So I'm doing that now.  Sorry, I forgot about that.

emilecantin, 2020-05-22:
There are _many_ philosophies on the "proper" way of branching; and it's a subject of continuous debate, mostly because everyone has slightly different requirements, so slight different definitions of "best".

_One_ way to do this is to use [git-flow](https://www.atlassian.com/git/tutorials/comparing-workflows/gitflow-workflow), and I've used it before on _very_ active projects, but I'd argue it's overkill for our needs. Unless you find yourself really needing it, I'd recommend keeping the `master` branch on the "bleeding edge" (all completed features / bugfixes), and having feature development / bug fixes in feature branches, which is already accomplished through the Pull Request workflow. More stable stuff would go in individual release branches / tags. The one thing that'd need to change in our existing workflow is that everyone, even maintainers, go through the Pull Request flow for all changes, except very trivial or urgent changes.

emilecantin, 2020-05-22:
Also, I've managed to build _something_ with pyinstaller, but I'm really not sure it's the right thing.

Care to share how you invoke it in your build scripts?

madgrizzle, 2020-05-22:
This is all I do to build the single directory version, and create a new release file.. single file is the same, just different spec file and different renaming (there's probably a way in the spec file to get things named the proper way from the start, but this worked nevertheless).

>pyinstaller main-onedir.spec
cd dist/main
mv main webcontrol
cd ..
mv main webcontrol
cd webcontrol
touch webcontrol-rpi-singledirectory.tar.gz
tar -zcvf webcontrol-rpi-singledirectory.tar.gz --exclude=webcontrol-rpi-singledirectory.tar.gz .

emilecantin, 2020-05-22:
Yeah, I've tried with the `main-onedir.spec` file, and I managed to build something that starts, at least. I didn't need to rename the executable, though, it came out as `webcontrol` right out of the gate. The folder is still named `main`, though.

Anyway, I'm trying to set up Travis in #149, but I need you to approve access to the organization (you should have an email about that).

madgrizzle, 2020-05-22:
Done.. I think

emilecantin, 2020-05-22:
Yeah, it's active, but only repo admins can add it (annoyingly). Can you add it here: https://travis-ci.org/github/WebControlCNC/WebControl ? Thanks!

madgrizzle, 2020-05-22:
I logged in but says "You don't have sufficient rights to enable this repo on Travis.
Please contact the admin to enable it or to receive admin rights yourself."

emilecantin, 2020-05-22:
It's telling me the same thing, weird.

madgrizzle, 2020-05-22:
Well, I went in and under my settings, I clicked on the webcontrolcnc organization and it showed two repos with switches.. i enabled the switches, but when I clicked settings, it says takes me to the travis repo and pops up an error message "Oh No! Your permissions are insufficient to access this repository's settings"

emilecantin, 2020-05-22:
Okay, I don't know what changed, but I refreshed the page and I was able to trigger a build. Success!

madgrizzle, 2020-05-22:
So, can this build win32 files?  I assume arm (rpi) is not feasible.

zaneclaes, 2020-05-22:
Fwiw, I run a BuildKite cluster already. It should be able to handle CI
needs and I have plenty of experience implementing it..., if Travis becomes
too much of a headache lmk.

emilecantin, 2020-05-23:
@madgrizzle Win32 I'm not sure, but Win64 will very probably work, as well as macOS. I'm starting with the easiest part, Linux on amd64. Arm will probably work as well, depending on the specific version we need; I'll test that once I have the whole pipeline working with regular Linux.

emilecantin, 2020-05-23:
[image]
Hey, looks like it's working! Off to test ARM now.

emilecantin, 2020-05-23:
Yeah, looking into this, having to compile the firmware with the release is an additional complication; it'd be way easier if I could just download the binary firmware releases directly. @madgrizzle I opened a pull request on your firmware repo do do this; it'll need some configuration on the Travis side as well, which I can't do myself (as it's not in the WebControl org).

madgrizzle, 2020-05-23:
If I move the repo here, would that solve it?

emilecantin, 2020-05-23:
We'll need to do the same little dance we did earlier today to enable Travis on the repo as well (altough I'm not exactly sure what we did).

emilecantin, 2020-05-23:
And while I'm looking into it, it looks like we're building a pretty old version for the stock firmware (jan 2019); there are some commits on master from this week...

madgrizzle, 2020-05-23:
My firmware repo has been moved.  I 'enabled' it in Travis
[image]

emilecantin, 2020-05-23:
Can you also set the same permissions for this one, please?

[image]

madgrizzle, 2020-05-23:
We are building specific points.. There's lots of commits that have been made, but none have made it into an official release.  Bar hasn't built a release since Jan '19.  

These are the sha's for the branches being built:

holey: 950fb23396171cbd456c2d4149455cc45f5e6bc3 (madgrizzle/Firmware - holey branch)
stock: e1e0d020fff1f4f7c6b403a26a85a16546b7e15b (MaslowCNC/Firmware - master branch)
custom: bf4350ffd9bc154832505fc0125abd2c4c04dba7 (madgrizzle/Firmware - don't recall precisely branch)

I think the sha's represent a point in time as I believe they change after a commit?

We can do away with the custom in reality since I've stopped work on it.

madgrizzle, 2020-05-23:
Doh.. that explains why still having issues with fake_servo.  I didn't update the sha for the latest commit.  holey should be:  e9cc445cfdfddcee4da97617e9d407cdc52d29b7

emilecantin, 2020-05-23:
Yes, I saw that. I think the approach I'm going to take to build these different versions is the following:

- Merge the changes from upstream (and try to keep up from time to time in the future)
- Create "release" branches on these 3 specific commits
- Build these 3 branches in the Travis build, and upload the hex files in an actual release

Then I'll able to simply fetch the latest release (or even a specific one) when building WebControl.

madgrizzle, 2020-05-23:
and stock is based on Jan 23, 2019 commit.. wee-bit behind.

emilecantin, 2020-05-23:
[image]
No, we're using the actual commit on which the release is based.

madgrizzle, 2020-05-23:
don't worry about the custom branch.. just build what's there until we pull it out of webcontrol entirely.

I still want to see a unified firmware though :)

emilecantin, 2020-05-23:
It's really no additional effort to include that one; I'll need write access to the repo, though. It doesn't seem to have the same permissions as the WebControl one.

madgrizzle, 2020-05-23:
Should be fixed now.

emilecantin, 2020-05-23:
Okay, we should now be synced up with that repo. `master` on the firmware repo should always track `master` on maslowcnc/firmware; that'll make it way easier. I moved your previous changes on master to the `develop/madgrizzle` branch, if you're looking for it.

emilecantin, 2020-05-23:
@madgrizzle if you're still here, I have one issue: I can't manage the build settings anymore, but I need to add a Github token in order to create a release from the build. (it can wait, it's getting late around here anyway).

[image]

madgrizzle, 2020-05-23:
Late here as well (east coast) you may need to talk me through it.  I'm going out of town tomorrow, but not sure what time I'm leaving or (after arriving) when I'll be able to get on my laptop.

emilecantin, 2020-05-23:
East coast here as well! Québec, Canada.

Go [here](https://github.com/settings/tokens), and create a token with the following permissions: `repo, write:packages` (`read:packages` will be added automatically). Once saved, you'll get back to the list of tokens, and the newly created one will appear in green. There's a long hexadecimal string, that's the token. Copy it.

Now go [here](https://travis-ci.org/github/WebControlCNC/Firmware/settings), and in the "Environment Variables" section, add one with the name `GITHUB_TOKEN` and paste your token as the value. Click "add", and it should be good. Let me know when it's done, and I'll trigger another build to test.

madgrizzle, 2020-05-23:
Done.. why does this need to be done for the firmware, but not the other two?

madgrizzle, 2020-05-23:
And how did I manage to 'assign' this to @zaneclaes ?  I don't recall doing so..

zaneclaes, 2020-05-23:
Haha, I was wondering...

emilecantin, 2020-05-23:
Not sure, but I lost access to Webcontrol as well. I managed to add the
environment variable when I created the thing on the Travis side, but when
you did something with the organization, I lost access to the settings. I
guess I probably have different permissions on the pi image repo, so I do
have access to that one.

emilecantin, 2020-05-23:
And it looks like creating a release works properly, and I get the 3 different firmwares!

Just note that the git setup is a bit convoluted at the moment; we'll be able to drastically simplify this when we get the holey firmware merged in.

[image]

madgrizzle, 2020-05-23:
Amazing work! So much better than my scripts!  Thank you so much for working on this.

gb0101010101, 2020-05-25:
Yes, great work. Really happy to see so much activity and progress.

I like the idea of using GitFlow as it corresponds with other workflow I have used before (dev, master, hotfix, feature etc). My only concern is if it makes things harder for the hobby contributor:

- Does it complicate the simple GitHub public Fork, PR process beyond telling contributors to use 'Develop' branch?
- Does everyone need to use GitFlow to submit a Feature PR? (looks like its not mandatory)

It looks pretty straight forward to use.

Does GitFlow integrate with or extend GitHub GUI to make project management easier?

emilecantin, 2020-05-25:
There are a lot of ways we can go about this, but if we want to go towards git-flow, we have to decide a few things:

- What should live in the default branch (`HEAD` on github)? i.e what should people get when they clone our repo? Stable code, or more bleeding-edge code?
  - This will guide whether we have `master`=stable, `develop`=bleeding-edge, OR `release`=stable and `master`=bleeding edge.
- Do we maintain old releases? Or do we expect users to update to the latest release to get all bugfixes?
  - This will guide whether we need the whole hotfix part of gitflow (probably not, IMO)

As for whether people need to use it to contribute, the answer is probably not, unless we want to maintain a clean git history tree. With the PR process that's already in place, we basically already have feature branches.

In my opinion, it's a little too much work for not enough benefit to maintain 2 main branches on this repo. With most people here not having a professional software development background, I feel it's better keeping things simple. However, I'd definitely encourage everyone to use git-flow conventions when naming their branches.

zaneclaes, 2020-05-25:
My two cents:

The default (`master`) branch should always be stable. To do otherwise amounts to building a foot-gun for unsuspecting Fork-ers to shoot themselves with. It's best to engender a good first experience, as it means fewer issues and more good-will.

I'd also argue for not maintaining releases, if only as a matter of expedience for a small team. The vast majority of software development hours, I've found, go toward maintenance. Until they can justify such effort (_read: $_), development teams have every right to insist users upgrade to the latest if they want to be supported.

Speaking of backgrounds... by way of introduction, mine is that I was most recently a software engineer (3 years) and then engineering manager (3 years) for Airbnb. I worked first on the mobile apps, then on the API, and finally on DevOps & Observability.

madgrizzle, 2020-05-25:
A lot of this is new to me, but what I envisioned that that we have a "stable" release and then have a series of development releases that we can test things out in.. once we are happy with what's in a development release, we do a PR and push those changes into the stable branch and create a new stable release.  

As for maintenance of releases, it seems like a lot of work to go back and update releases to fix issues.. I think we'd have to have a major change to think about doing that.. like a WebControl 2.0 that works differently and some people just might want to stay on WebControl 1.0.

I also think new features need to be added in a manner to maintain backwards compatibility. For example, the support for R in g2/g3 and the zaxis limits should still work for people using old firmware. Doing it this way, I think, would allow people that just want stable, reliable operation to know that anytime they upgrade, it's not going to break their installation and they can then chose to use new features or not.  This way they can get but fixes, but not be forced into making any other changes.

zaneclaes, 2020-05-25:
Perhaps a stupid question... who are the maintainers of GroundControl? To what degree do we expect feature parity between the two? 

Personally, I feel that it would be better to offer the community a single piece of software. WebControl IMO is the better approach for a number of reasons, especially: (1) OS-agnostic (2) web UI development is significantly faster than native/client development (3) remote access & monitoring is a killer feature. With some investment in modern UI frameworks and some cleanup all around, I could see WebControl becoming very user-friendly and feature-rich application.

_psst... kinda wish it had been called AirControl... or AirCNC... but that's just because I like puns..._

zaneclaes, 2020-05-25:
Imagine: a built-in asset catalog that lets you upload your own designs and explore others' designs from within the app...

[12 more replies]